guysoft/OctoPi #143: Wifi Config Not Working As Described

xentricag04, 2015-09-05:
Editing the /boot/octopi-network.txt file (per instructions on OctoPrint website) did not work.  Had to manually input those settings into the /etc/network/interfaces file.  I suspect the line "source /boot/octopi-network.txt" in the interfaces file was not working as intended for some reason because I literally copied the text I had placed in the octopi-network.txt file into the wlan0 section of the interfaces file, commented out the "source..." line and it worked perfectly.

foosel, 2015-09-05:
I've configured the WiFi this way now for around 25 or more flashs over the last weeks. It worked every single time.

foosel, 2015-09-05:
Meh, hit the button to submit accidentally (tablet). 

Having stated the above, my guess would be it's either something that went wrong when you entered it into the file the first time or it has something to do with your particular WiFi setup. Any examples or other additional information you could provide? Otherwise this pretty much boils down to "works for me and others, doesn't work for you", that would be an unsatisfying outcome.

xentricag04, 2015-09-05:
For reference, the build I flashed was 2015-05-05-octopi-wheezy-0.12.0.img

I uncommented the 2 WPA/WPA2 lines and filled in my SSID and password and saved.  Several boot cycles, no joy.

Double-checked all the spelling.  All correct.

Copied and pasted the 2 lines into the other file.  Commented out the wlan1 section (not needed in my case) and the “source…” line and saved.  Worked perfectly after that.

I totally recognize it could be a noob error on my part as this is only my second flash, but the wifi setting sections of each file are line-for-line identical.

xentricag04, 2015-09-05:
Correction to my last.

I had uncommented the 3 lines in the octopi-netowork.txt file that were applicable.  The following 3 lines are what I copied into the interfaces file.  They were identical between the two files per my previous description.

iface wlan0 inet manual
    wpa-ssid “*****”
    wpa-psk “*****”

foosel, 2015-09-05:
Exact same config version and same lines work great in all my tests (with my WiFi credentials of course), when included like described in the readme. Weird. 

Can you still reproduce this or was that a "one time thing"?

xentricag04, 2015-09-05:
I'll reflash from scratch and replicate my original steps to see if I can reproduce.  I'll let you know in about 30 if the published procedures worked the second time around.

xentricag04, 2015-09-05:
Could not reproduce, but I did get a bad flash and had to do another to get it to work.  Must be poor SD reader I'm using to flash.  Therefore, I can only assume the first flash must've been bad (i.e. missing some dependency files).  I will close the issue.  Sorry for the chaff.

closed 2015-09-05 by xentricag04