guysoft/OctoPi #145: OctoPi not reconnecting Wi-Fi when network goes down

Nurgak, 2015-09-08:
I had some issue with my modem/router lately and it needed a restart. I was printing using OctoPi (1.2.6) and I'm using a Wi-Fi dongle to communicate with it. When the network was up again OctoPi did not reconnect (the print was OK though). The only way to get it to reconnect, without power cycling the Pi, was to unplug and replug the Wi-Fi dongle.

I have another RPi running [OSMC](https://osmc.tv/), which behaved correctly and reconnected immediately, with the same type of Wi-Fi dongle.

It seems a connection drop and reconnect is [not](http://www.lynsayshepherd.com/blog/2014/06/27/keeping-the-raspberry-pi-connected-via-wi-fi/) [something](http://itsbrent.net/2014/05/fixing-raspberry-pi-wifi-dropping/) [trivial](http://unix.stackexchange.com/questions/224881/wi-fi-drops-its-really-complicated-to-autoreconnect-raspberry-pi-with-octopi). So I prefer asking here for the best way to handle this for OctoPi.

digas, 2015-09-08:
In /etc/network/interfaces
If you don't have this lines please add it :

auto wlan0
    allow-hotplug wlan0

guysoft, 2015-10-05:
Please confirm this worked, otherwise I'll close the issue for inactivity

Nurgak, 2015-10-11:
I added the lines and tested it by hard resetting the modem/router, but it still doesn't reconnect.

MolarAmbiguity, 2015-10-11:
Those lines are present by default (at least in 2015-05-05-octopi-wheezy-0.12.0), and it doesn't reconnect.

Nurgak, 2015-10-29:
This did the job: http://alexba.in/blog/2015/01/14/automatically-reconnecting-wifi-on-a-raspberrypi/

It's pinging 8.8.8.8 every 5 minutes and soft-restarting the Wi-Fi in case it doesn't have a response. This is good enough for me.

guysoft, 2015-10-29:
This is pretty cool, @foosel do you think we could add something like that?
It will need detection of active network interface.
@Nurgak Any other recommendations?

foosel, 2015-10-29:
I'd vote against something that depends on _internet_ connectivity to detect if the wifi device is still connected to the _local_ network - what if you run your printer in an environment where it doesn't have internet access? OctoPi would become completely unusable then. 

Pinging the default gateway from `ip route` might be interesting though:

```
oprint pi@octopi ~ $ ip route
default via 192.168.1.1 dev wlan0  metric 303
192.168.1.0/24 dev wlan0  proto kernel  scope link  src 192.168.1.160  metric 303
```

Might also be interesting to look if the output of `/sbin/ip neigh show` can't be utilized here - that's what I used in `netconnectd` to detect if a network link is established at all

```
oprint pi@octopi ~ $ ip neigh show
192.168.1.1 dev wlan0 lladdr bc:ee:7b:79:73:b8 STALE
192.168.1.3 dev wlan0 lladdr 14:da:e9:0f:87:5b DELAY
192.168.1.10 dev wlan0 lladdr 00:11:32:11:48:b3 STALE
192.168.1.2 dev wlan0 lladdr bc:ae:c5:18:c9:0a STALE
```

Filtering that by the monitored device (`wlan0` unless otherwise specified) and then checking if there are lines that don't end in "INCOMPLETE", "FAILED" or "NONE" might work here. But that's only a guess for now, can't test that at the moment.

PR #174 (2015-12-13): Auto reconnect WLAN after a lost connection
This is my attempt to fix #145, it uses a script from the wpa_supplicant package.
closed 2015-12-13 by guysoft