guysoft/OctoPi #146: Error: Failed to autodetect serial port, please set it manually.

Balkmeister, 2015-09-30:
Hello, i recently tried to get my zeepro zim working with OctoPi and everything went smooth right up untill i tried to connect to the printer. i had selected Auto serial port, 25000 baud rate and Default printer profile and kept getting this message:

Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: []
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.

I then tried to add /dev/ttyAMA* to the list of serial ports manually using the "Additional serial ports
" box in the serial connection settings and instead got this:

Changing monitoring state from 'Offline' to 'Detecting serial port'
Serial port list: ['/dev/ttyAMA0']
Connecting to: /dev/ttyAMA0
Error while connecting to /dev/ttyAMA0: 'Timeout'
Changing monitoring state from 'Detecting serial port' to 'Error: Failed to autodetect serial port, please set it manually.'
Failed to autodetect serial port, please set it manually.

The time between pressing connect and getting the error message cant have been more than a second so not sure how it could timeout exactly. I have tried two usb cables but that didnt help. I also tried switching ports (i have an usb cam connected to one port and an usp cable to my printer in the other) and while the camera worked in either port i never could get the pi to connect to the printer.

Any ideas?

nophead, 2015-09-30:
/dev/ttyAMA0 is the TTL serial port on the GPIO connector, not a USB serial port.

foosel, 2015-09-30:
First of all, this isn't a problem with OctoPi (the sd card image) but rather with the communication between OctoPrint (the print server you are looking at) and your printer, so a better place for this would be [the OctoPrint Mailinglist](https://groups.google.com/group/octoprint) or [the OctoPrint G+ Community](https://plus.google.com/communities/102771308349328485741).

That being said, since you said that you are connecting your printer through USB, `/dev/ttyAMA0` is  definitely the wrong port - that is actually the serial port of the Raspberry Pi itself, which by default is serving a serial console which would explain why you can't connect to it - it's not your printer. That's also the reason why it's not part of the default ports to allow connecting to btw ;)

OctoPrint by default assumes your printer to be available under one of these devices:

``` python
	baselist = baselist \
			   + glob.glob("/dev/ttyUSB*") \
			   + glob.glob("/dev/ttyACM*") \
			   + glob.glob("/dev/tty.usb*") \
			   + glob.glob("/dev/cu.*") \
			   + glob.glob("/dev/cuaU*") \
			   + glob.glob("/dev/rfcomm*")
```

If none of those match your printer, it can't connect unless your specify the proper device your printer registers under in the additional ports field. Most printers do NOT need that however, so I suggest to first figure out if OctoPi/Linux is detecting the printer in the first place - disconnect it, connect it, on a terminal (monitor + keyboard or SSH, default username is "pi", default password is "raspberry") write "dmesg" and look for anything hinting at a new USB serial device having been detected. 

It might be helpful to know what kind of printer it is, some need special drivers.

Balkmeister, 2015-09-30:
Thanks guys for helping me even though it wasnt an octopi problem. It turns out one of the pins on my printers usb connection wasn't getting a physical connection with the cable and since i'm a total linux noob i assumed it was due to software user error. After i fixed the connector everything connected just fine =)

closed 2015-09-30 by Balkmeister

jcernato, 2015-10-15:
Hi!
I'm experiencing similar issues. I upgraded octoprint from 1.1.1.14 to 1.2.6 today and now I can't connect to my printer anymore.
My printer is connected via ttyAMA0 and a level shifter! - Baudrate: 115200 
I've put the "/dev/ttyAMA0" into "additional ports" but it tells me "timeout" right after clicking "connect"...

> Changing monitoring state from 'Offline' to 'Detecting serial port'
> Serial port list: ['/dev/ttyAMA0']
> Connecting to: /dev/ttyAMA0
> Error while connecting to /dev/ttyAMA0: 'Timeout'
> Changing [...]

If I set up a serial connection via command-line (ec. minicom) the printer responds fine.
Disconnecting the serial connection and plugging in the USB-cable (ttyACM0) works fine with octoprint aswell...

guysoft, 2015-10-15:
@foosel When you get back to the office next week.

guysoft, 2015-10-15:
@jcernato ttyAMA0 is definitely the wrong port, read @foosel reply.

jcernato, 2015-10-15:
[OctoPrint-Change-Log](https://github.com/foosel/OctoPrint/blob/master/CHANGELOG.md)
Section 1.2.0 (2015-06-25)-Improvements:
>Serial ports matching _/dev/ttyAMA*_ are not anymore listed by default (this was the reason for a lot of people >running into problems while attempting to connect to their printer on their Raspberry Pis, on which _/dev/ttyAMA0_ is the OS's serial console by default). Added configuration of additional ports to the Serial Connection section in the Settings to make it easier for those people who do indeed have their printer connected to _/dev/ttyAMA0_.

Once again, I do have connected my printer via ttyAMA0 (raspberry Pi GPIO header), it worked perfectly fine with the old octoprint-version and it still works if I send the commands manually via command-line to ttyAMA0.

nophead, 2015-10-15:
Yes I also use ttyAMA0 and some resistors to change the levels and connect
it the second serial port on a Melzi. It definitely works on older
versions. I haven't tried 1.2.

foosel, 2015-10-15:
@jcernato This has nothing to do with OctoPi (the sd card image) but OctoPrint (the print server). Please [open a bug report on the OctoPrint bug tracker](https://github.com/foosel/OctoPrint/wiki/How-to-file-a-bug-report), make sure to include `octoprint.log` **and** `serial.log` (which you'll need to enable to be written) and to use the bug reporting template. I'm out of office until monday, but there are other people who might be able to help over in the correct bug tracker.

That being said, I know for a fact that 1.2.6 works perfectly fine in general with `/dev/ttyAMA0` since I run a printer myself that way.

jcernato, 2015-10-16:
As a colleague of mine used to say: "electronics have a soul", this seems to apply to software too.
Today I made a fresh start and at first it didn't give me a timeout error, it just closed the connection after a few seconds without getting a response from the printer.
After enabling the serial.log it is now working. I have absolutely no idea why but I'm glad it does.

I thank all of you anyway for the help and quick responses, and mostly for developing octoprint!

CJAndrade, 2015-11-21:
@jcernato Thanks for you post , after trying various things and flashing the SD card a couple of times this week. I went with you suggestion of enabling/checking  flag for Log communication to serial.log (might negatively impact performance),  after upgrading to 1.2.6 from the default version which comes with 2015-07-02_2015-05-05-octopi-wheezy-0.12.0.zip..

thebabymaker, 2016-01-17:
Does octopi support the dremel idea builder? I've set it up but can't get it to connect. I've checked dmesg and the rpi detects and recognizes the printer. I can provide more details but I just wanted to first check if there is a known compatibility issue with the dremel. Thanks!

markwal, 2016-01-18:
@thebabymaker I don't think that will work.  The Dremel Idea Builder is, if my information is correct, a whitelabelled FlashForge Dreamer.  The Dreamer went closed source (the Creator X uses the open source firmware Sailfish) and I think went to a new microcontroller (ARM based maybe?) and no longer uses the x3g protocol.  So OctoPrint can't talk to it either by itself or with the OctoPrint-GPX plugin.

chapimp, 2016-02-23:
I have a PowerSpec Ultra 3D  [Flash Forge Dreamer clone]  from Microcenter.
Would I be able to install Octoprint on it.
I've googled it but no luck.

markwal, 2016-02-23:
@chapimp No.  Not likely.

The PowerSpec Pro would work because it is a clone of the FlashForge Creator Pro which is a clone of the MakerBot Replicator 1 Dual which was open source.  The Ultra on the other hand is a FlashForge Dreamer which is closed source and has a proprietary version of firmware and protocol and it is poorly documented so it seems doubtful that it would work.

It's unclear how different the new stuff is so there's a chance that a few fixes would get it working but it'd be some significant black box testing and iteration to try I'm sure. If you're a dev, you'd need to start with a line capture of what goes over the USB from the print software that works and compare that to known protocols that they may have borrowed from (probably x3g).

pelitt, 2017-07-29:
Hi my pc is orangepilite and Armbian Last software (30 desktop Edition ). 
i install octoprint but give me this error. I want try edit ttyAMA0 file but i dont have this file under dev folder.  İt have  tty0,tty1,tty2... And this files not editable whit  sudo nano code.please help me

pelitt, 2017-07-30:
hi i set Serial Port that /dev/ttyS0
Does not give an error but is not connected

markwal, 2017-07-30:
@pelitt you can't edit those files. They're psuedo files. When a program writes to them the bytes go out and reading just reads the sequence of bytes accumulated from the wire.

markwal, 2017-07-30:
ttyS0 is unlikely to be your printer

pelitt, 2017-07-30:
@markwal I saw ttyS0 in the result of serial connection. So I added ttyS0 to the Additional serial ports list in octoprint settings 
[image]

in this way
[image]

pelitt, 2017-07-30:
@markwal Tell me what I have to do? How can i solve this problem?

markwal, 2017-07-30:
That ttyS0 looks like it is the port your console is connected to (ie it's in the login prompt above) *not* your printer.  Probably best to continue this on the forum/community.

pelitt, 2017-07-30:
@markwal thanks this when octoprint server is test log screen
[image]