guysoft/OctoPi #147: OctoPi failing to build

MolarAmbiguity, 2015-10-01:
Hi, I'm trying to build OctoPi on my Laptop (Linux Mint 17.2 32-bit). The build is failing when it gets to unpacking the filesystem.  Here's the output: https://gist.github.com/MolarAmbiguity/2b22eadff703625a516a
Something seems to be up with qemu (maybe related to me being on 32-bit?)

guysoft, 2015-10-05:
That error means this line did not work:
```+++ sed -i 's@/usr/lib/arm-linux-gnueabihf/libcofi_rpi.so@\#/usr/lib/arm-linux-gnueabihf/libcofi_rpi.so@' etc/ld.so.preload```
. I would look at the ``etc/ld.so.preload`` file and see if that line is commented out.
Are you using raspbian or something else?

There is some so file that gets automatically loaded that if its loaded in the chroot gives the illegal instruction error.
This should still work if you are on 32bit, the qemu-arm-static that is downloaded should be 32bit.

MolarAmbiguity, 2015-10-09:
> Are you using raspbian or something else?

I'm adding octopi to raspbian

> I would look at the etc/ld.so.preload file and see if that line is commented out.

I cannot see that file in the octopi filesystem directory or in my own /etc folder.

guysoft, 2015-10-09:
What version of raspbian?

That file should exist in the chroot, of its not then likely the line ot contained moved somewhere else. I would run a "grep -r" search in the chroot etc folder and see if its located somewhere else, remember you need to escape "/" in grep.

MolarAmbiguity, 2015-10-09:
> What version of raspbian?
2015-09-24-raspbian-jessie

I can't find ld.so.preload. At what point should that file have been created? It isn't present in the git repo?

MolarAmbiguity, 2015-10-11:
Should I be searching for ld.so.preload during the build? when the image is mounted?

guysoft, 2015-10-12:
See https://github.com/guysoft/OctoPi/issues/149
For now use the wheezy image, or help us port it :)

MolarAmbiguity, 2015-10-20:
Thanks for the help @guysoft, everything worked fine with the wheezy image.

I currently don't think I have the skills or experience required to help port to jessie. If you could point me towards any useful resources I'd be grateful.

guysoft, 2015-10-20:
@MolarAmbiguity I pointed you to the code that needs to be changed, if you don't have the skills its kinda hard to now where to start. I guess we will need to grep for the ``libcofi_rpi.so`` location in ``/etc``. Unless you feel adventures best wait. Its not like upgrading Debian version is limiting OctoPi. Its just something we need to do so future revisions work.
Are you building OctoPi for a specific reason?

guysoft, 2015-10-27:
@MolarAmbiguity Ok fixed this preload issue in the jessie branch, will push that soon to devel, closing for now. If anyone needs the commit is: https://github.com/guysoft/OctoPi/commit/eee2ad9c3a99164950e356b5e852338c3f59a005

closed 2015-10-27 by guysoft