guysoft/OctoPi #149: Make OctoPi work on Raspbian Jessie

guysoft, 2015-10-10:
Raspbian has released on September 25 a build using Jessie.
It seems it does not build out of the box using the current scripts.

Would be helpful if anyone comments here what is not working, and solutions.

This issue is also to decide what bloatware we can remove, since they added yet more applications.

Release notes:
```
2015-09-25:
  * Based on Debian Jessie
  * Upgraded applications - Epiphany browser, Scratch and Sonic Pi
  * Included applications - LibreOffice, Claws Mail, Greenfoot, BlueJ
  * Included utilities - Alacarte menu editor, Lxkeymap, scrot, tree, pip
  * New GUI-based Raspberry Pi Configuration application
  * GPIO control now possible without need for sudo
  * Web link to Magpi magazine included
  * New taskbar plugin to eject mounted USB drives
  * Default boot is now to GUI not desktop
  * Look and feel now based on GTK+3 default theme
  * Print screen key launches scrot to produce screenshot
  * Common keyboards autodetected by GUI and drivers loaded accordingly
  * Numerous small tweaks and bugfixes
```
Taken from:
https://downloads.raspberrypi.org/raspbian/release_notes.txt

foosel, 2015-10-10:
They hinted at a light image being in the works in the comments of the [release post](https://www.raspberrypi.org/blog/raspbian-jessie-is-here/):

> It might not be exactly what you’re after, but a lite image is coming soon. It won’t have any of the bundled additional software, and won’t have X.

Sounds like the perfect starting point IMHO, so I suggest waiting until that light image is available before adapting, or the work might have to be done twice.

seefood, 2015-10-11:
I'll +1 the light image approach either way. also, this needs an upgrade path. I still haven't understood why the Octopi is an image and not a deb source with 5-6 deb packages.

foosel, 2015-10-11:
Probably because nobody so far packaged all that stuff that makes it tick and shared the results here :)

In any case, there would still need to be a "plug-and-play" like image. From the support requests I see on the internet, adding an additional apt-source to a regular raspbian and then apt-get'ing the server setup that way would still be too much for a lot of people out there.

seefood, 2015-10-11:
Well, so there will be a snapshot once in a few months, and the rest of the time regular deb  upgrades. it will improve testability, upgradabilty and make the daily build much smaller...

guysoft, 2015-10-12:
There seem to be so preload issues, I'd guess they moved the line to a different file
https://github.com/guysoft/OctoPi/issues/147

commit eee2ad9c (2015-11-03): Get build to work on Debian Jessie fixes #149
closed 2015-11-03 by guysoft