guysoft/OctoPi #151: Automatic start Octopi

krzysiekd, 2015-10-13:
I managed to implement Octopi on Raspberry pi B+ but I have small problem.
I find a script for Automatic start Octoprint. Of course it's a script for Octoprint,not for Octopi (I understand that pi version it's minimized normal version). It's possible to use this script in Octopi ?

guysoft, 2015-10-13:
I am not sure I understand your English, OctoPi has a startup service set up that starts up OctoPrint.
Its located at ``/etc/init.d/octoprint``.
The source is here: https://github.com/guysoft/OctoPi/blob/devel/src/filesystem/root/etc/init.d/octoprint

If this is not what you want then try and explain what you wanted better.

guysoft, 2015-10-29:
Closing due to inactivity unless someone objects

closed 2015-10-30 by guysoft