guysoft/OctoPi #152: How do I change the webserver port Octopi runs on?

bolsoncerrado, 2015-10-14:
ie. what if i wanna run Octopi on http://192.168.1.100:88 instead of default 80 port?

Thanks!

guysoft, 2015-10-14:
You can change the port haproxy is using, this source is here:
https://github.com/guysoft/OctoPi/blob/devel/src/filesystem/root/etc/haproxy/haproxy.cfg

The path on your pi woud be at:
``/etc/haproxy/haproxy.cfg``
Change the ``80`` there in ``frontend public``  to ``88``.

closed 2015-10-14 by guysoft

saintofinternet, 2018-01-23:
how can i make repetier-host communicate with ocotpi over wifi?? the port 8876 is getting refused.

guysoft, 2018-01-23:
@saintofinternet you are:
1. confusing the webserver port with the method the printer is communicating with OctoPrint
2. Confusing OctoPi for OctoPrint. Please post in OctoPrint's issue tracker