guysoft/OctoPi #153: unable to connect to wifi

jiaqiren, 2015-10-15:
I've been used 0.11 Octopi on my Pi2 for month with TP-Link WN725N V2. And recently, I decided to change to 0.12. However, the new DHCPCD seems not work with WN725N V2 very well. Completely lost my WIFI connection. I wonder is there anyway to get rid of defaut DHCPCD and back to WPA?

Thanks a lot!

markwal, 2015-10-15:
Weird.  I have a TP-Link WN725N (I think v1) and I have to manually install a driver for it to work win any version on OctoPi.  Does Raspbian Wheezy include a driver for the v2?

jiaqiren, 2015-10-15:
Yeah, obviously V2 has different chipset as V1. It took me quite time ti set up for 0.11 too....

AbuMaia, 2015-10-16:
I'm also having trouble, but I use an Edimax wifi dongle. It flashes a few times shortly after the Pi is booted, but it doesn't connect to my router. Even plugging the Pi into the router via an ethernet cable doesn't let it connect to the network. It never gets an IP address. This new "change in network configuration" has me baffled.

jiaqiren, 2015-10-17:
I just bought an Edimax 7811un dongle. After couple set up through command, it works!

AbuMaia, 2015-10-17:
I gave up on 0.12, and I'm using 0.11 with no problem. Just a quick edit of the wpa_supplicant.conf file got my wifi working first try.

guysoft, 2015-11-24:
You can try 0.13 nightly, that might solve it

cocoknight, 2015-12-14:
I do have the same issue even with the latest nightly. I'm using a Raspberry Pi 2 and an EDIMAX WiFi dongle (http://www.amazon.de/EDIMAX-EW-7811UN-Wireless-Adapter-IEEE802-11b/dp/B003MTTJOY/ref=sr_1_3?ie=UTF8&qid=1450122319&sr=8-3&keywords=wifi+dongle).

foosel, 2015-12-14:
Have the exact same one in use with several installs here, works flawlessly and every time. So it's probably not the hardware (unless the dongle is broken).

Make sure your text editor didn't mangle the quotation marks around ssid and password (apparently some text editors love to do that), that the comment # at the beginning of all three lines (iface, ssid, psk) has been removed and that the credentials are correct.

milkypostman, 2015-12-25:
i'm having this same issue. i'm not sure how to debug it.

n2b8, 2016-01-26:
I'm having the same issue as well. Is there any chance that having spaces in my SSID could be causing this?

GriffinPrice, 2016-02-25:
I'm having the same issue as cocoknight above (same dongle and everything). I get a few blinks on the dongle during start up and then nothing. I'm connecting to an unsecured network that whitelists MAC addresses. I've tested the dongle on another computer, and it connects fine, so I am confident that it is not a dongle hardware problem or external problem with the network. I've tried modifying the network config with three different editors at this point: nano, vim, and brackets.

I think there is something wrong with the service initialization, as I have no network utilities when I start up the GUI. Instead of the wifi signal shown [here](https://www.raspberrypi.org/blog/another-raspbian-desktop-user-interface-update/), I get a little icon that seems to mean "ethernet". `sudo dhcpcd` doesn't initialize any networking as far as I can tell, and I can't get any UI for connecting to access points using the guides I have seen thus far.

Can anyone give some guidance? Is `dhcpcd` the correct way to get networking going or is there another route? Can any superusers give me some commands to go through or diagnostic tools? I don't have access to the internet on my pi and I don't have control over the network in my location, so doing a wired connection to sort this all out is not really possible.

foosel, 2016-02-25:
If your ssid or password contain spaces, try single quotes instead of double quotes around them (`'my ssid'` instead of `"my ssid"`)

Also, please ignore the ui tools, they seem to cause more trouble than helping in any way.

Check that `iwconfig` and `ifconfig` see your device, cross check the reported mac address with the one that is white listed.

GriffinPrice, 2016-02-25:
I did have the ssid in double quotes, but changing to single quotes did not help. `iwconfig wlan0 essid 'my network'` is connecting the dongle after I've logged into the pi, but it is not connecting automatically and I have to connect a screen each time to get anything useful out of it. I think there should be a config file somewhere on the filesystem that I can fiddle with to get it to go automatically on boot, but I am not sure. When it works, it works pretty goddamn well.

foosel, 2016-02-25:
`octopi-network.txt` is supposed to do exactly that, you shouldn't need to have to trigger it manually. Maybe paste the output of `dmesg` and `cat /var/log/syslog` on pastebin.com and link to it here, there might be something in that that explains the troubles. I'm using the edimax dongles almost exclusively and on some days setup octopi multiple times for tests, including successful Wi-Fi configuration just by editing the above file once even before the first boot, so whatever it is, it is a specific problem.

GriffinPrice, 2016-02-26:
`cat /var/log/syslog` is too big to deal with (I left `cat /var/log/syslog > syslog.txt` running for an hour and it did not complete. [Here is a paste of `dmesg`](http://pastebin.com/7wzPZw7V). Do I need to do a fresh install each time I modify `octopi-network.txt`? I checked it in `//boot/octopi.txt` and everything looked fine. Just in case, I re-typed the network address and single quotes and saved using nano on the RPi. When that didn't work, I went back and did it again with double quotes.

As much as I hate it when people post pictures of their screens, I have decided I need to do it. [Here is an imgur album.](https://imgur.com/a/ouIrJ). First is a picture of my `octopi-network.txt`, opened on the pi. I know it has double quotes in this picture, but I **promise** that I tried it with single quotes as well. I've been trying thus far to preserve some level of anonymity by omitting my wifi network name, but I think I'm running out of options. The second picture is the boot console output, including around the time when networking was being done. There's some `dhcpcd` errors being thrown which seem like they shouldn't be there. 

Thanks for the ongoing help. I'm gonna do another fresh install and an octopi-network.txt write with a linux system while I wait for cleverer eyes to take a whack at this.

foosel, 2016-02-26:
I should have specified that syslog already is a file, but one you usually need super user rights (so `sudo`) to read. And it should definitely not be so big it takes hours to dump.

No, you don't need to reinstall after changes to octopi-network.txt. It's literally just included in `/etc/network/interfaces` through the `source` statement, so no magic is going on there.

`dmesg` at least verifies it sees your device on boot. The output during startup ("no carrier") hints at your Wi-Fi dongle not associating. What's logged in syslog might help here. The output of `iwconfig` and `ifconfig` might too. Might also be interesting to compare if the output of any of them changes when you switch between single and double quotes. [There exists some evidence that for ssids with spaces single quotes should be the right choice](http://askubuntu.com/questions/50876/how-to-connect-to-a-wireless-network-that-has-a-two-word-name-with-a-space) (and yes, we'll need to change octopi-network.txt after verifying that).

foosel, 2016-02-26:
TP Link needs additional drivers to work at all. Is the SSID hidden by any chance?

GriffinPrice, 2016-02-29:
Sorry for the delay, but I'm back with updates. [Here is a paste of syslog](http://pastebin.com/D3sujaL2). It is now manageably long because I did a reinstall, but the previous one had over a day's worth of lines with "connection failed" errors happening about 6 times per second, which gives a conservative estimate of somewhere around half a million useless lines in the file. (At 30 lines per second, this means cat would take a little less than 5 hours to finish).

[iwconfig here](http://pastebin.com/itiixwPP)

[ifconfig here](http://pastebin.com/SD6sz5cb)

SSID is not hidden. I have copies of iwconfig and ifconfig from when I changed the config to double quotes. diffs of the iwconfigs has "encryption key: off" in the version with single quotes. Diffs of the ifconfigs only have differences is terms of the packets received/trasmitted, nothing significant.

I'm gonna buy a different dongle and see what happens.

foosel, 2016-03-01:
I don't think a different dongle will solve this issue, especially not if the dongle associates just fine when you ask it to do so manually. This reeks more of an issue caused by software or the environment.

I have to admit, at this point I'm a bit at a loss. In the syslog it sees the dongle but then it doesn't even try connecting at all. I don't see anything hinting at it failing to associate with the AP. So either it doesn't even try (Just to make sure, it does see the AP in `iwlist wlan0 scan`?) or it tries, fails and for whatever reason it doesn't get logged.

What you could try is putting your `iwconfig essid` line into `/etc/rc.local` (*before* the exit 0), that should make it run on startup, maybe get you going, but it would be a pretty ugly way of getting stuff to work that is supposed to be handled by the network sub system on its own.

What does happen if you do `ifup wlan0`? 

Also, just to verify, this is an unmodified install of OctoPi 0.12 and you did only change `octopi-network.txt`, right?

GriffinPrice, 2016-03-01:
To the last point, yes, this is an unmodified install of Octopi, fresh from the download page, with the hash verified. I've only fiddled with `octopi-network.txt` and run the other commands mentioned here. My pi is a Raspberry Pi 2 Model B V1.1.

`ifup wlan0` gives `interface wlan0 already configured`

My network is the first to appear in `iwlist wlan0 scan`, but there is something tricky going on here: it is a network with multiple access points. Could it be that the boot code doesn't play well with a multiple AP wireless network? At this point, that's my best guess.

foosel, 2016-03-01:
Would surprise me. Can't remember testing this specific setup with OctoPi,
but I'm fairly sure I've used it with regular Linux systems in the past,
without having to define a specific ap to use or something like that. Just
simple entries in /etc/network/interfaces (which is what inches
octopi-network.txt), like here.

Also sorry, try

    ifdown wlan0 && ifup wlan0

What I'm trying to see is what happens when it tries to bring up the
interface as defined in the configuration. Something must be going wrong
here causing your Wi-Fi to not get a carrier, but it doesn't leave anything
in the log as far as I see, so maybe triggering that manually will give
some insight. Hopefully.

milkypostman, 2016-03-02:
Let me put my anecdote here: my wifi failed because I edited the wifi settings file on my Mac using the default text editor. Apparently that editor uses a weird quotation so " was not the ascii quote but some weird Unicode version. Please check that this is not your problem as it was mine and the password was wrapped incorrectly. 

I only found this out through frustration where I hooked up a monitor and tty could not render the quote. Even ssh deceived me since my terminal on my Mac could render the weird quotations and they look almost identical. 

Once I fixed the quote problem, by copying them from other parts of the file, the wifi has been *rock solid*.

guysoft, 2016-05-01:
Closed, the issue is for the mac problems is at:  #222

closed 2016-05-01 by guysoft