guysoft/OctoPi #155: Plugin isntallation fails on jessie

raminger, 2015-11-04:
With the latest nightly (2015-11-03T19:55:03.702Z) the first try to install a plugin via the plugin manager fails.
The second try is successfull.

See here for pip.log:
https://gist.github.com/34e52962c458154fd626.git

foosel, 2015-11-05:
Hm, on first look this looks more like a hiccup in your network connection or github's servers than anything related to OctoPi or OctoPrint. See http://pyopenssl.readthedocs.org/en/latest/api/ssl.html#OpenSSL.SSL.ZeroReturnError

Does this happen every time or did it only happen once?

guysoft, 2015-11-05:
Indeed unless someone can confirm its reproducible it will be closed

raminger, 2015-11-05:
I setup a new OctoPi 0.12 installation, updated it to OctoPrint 1.2.7, and tried to install the Pushbullet plugin in the plugin manager. The installation of the plugin worked as expected, no ZeroReturnError on wheezy.
With a new clean OctoPi 0.13 installation I get the ZeroReturnError again...

So, imo this does not look like a network or server related issue.

Do you have any advice, how I can track it further down?

WheresWaldo, 2015-11-19:
I just tried this from this mornings nightly and confirm that it requires two installs to add a plug in. The first one fails with "unknown URL" (assmuing a generic error message here), and the second succeeds.

guysoft, 2015-11-24:
Is anyone still getting this ?

WheresWaldo, 2015-11-24:
Did something change with the nightly builds in the last 5 days. If so, I can test it over the weekend.

guysoft, 2015-11-25:
Not to my knowledge.
What do you mean 'plug'?

WheresWaldo, 2015-11-25:
Sorry, too quick to edit I meant I installed a plugin, using the Plugin
Manager. As already mentioned, the first installation almost always fails,
then you click install a second time and PIP successfully installs the very
same plugin. Like I said in my post, the only plugin I personally use that
installed correctly on the first try was the Status Line Plugin.

guysoft, 2015-11-25:
@foosel I now that github clones had issues in the build (when running from the pi chroot), thats why I use mirrors for the nightly builds, because they keep getting stuck. Perhaps this happends also on running Pis?

foosel, 2015-11-25:
I'm rather leaning towards a change in how the pip version on jessie handles things (e.g. ssl checks against pypi). I've been meaning to take another look at this, but I haven't gotten around to it yet.

For a start, if anyone could tell me what `pip --version` outputs, that would be great - as much as it pains me to say, I haven't even come around yet to fire up one of the new images myself :(

WheresWaldo, 2015-11-25:
pi@octopi ~ $ pip --version
pip 1.5.6 from /usr/lib/python2.7/dist-packages (python 2.7)
pi@octopi ~ $ lsb_release -a
No LSB modules are available.
Distributor ID: Raspbian
Description:    Raspbian GNU/Linux 8.0 (jessie)
Release:        8.0
Codename:       jessie
pi@octopi ~ $ cat /proc/version
Linux version 4.1.13-v7+ (dc4@dc4-XPS13-9333) (gcc version 4.8.3 20140303 (prerelease) (crosstool-NG linaro-1.13.1+bzr2650 - Linaro GCC 2014.03) ) #826 SMP PREEMPT Fri Nov 13 20:19:03 GMT 2015
pi@octopi ~ $

Does this info help?

foosel, 2015-11-25:
Ok, I downloaded the image and tried my hand at it. It's indeed an error in the included `urllib3` version when working with pyopenssl (which jessie now seems to ship with finally). `ZeroReturnError` is an exception signalling that the SSL connection was closed *cleanly*, urllib3 1.9.1 however doesn't handle that properly and throws an error. At that point the data has already been cached on disk however, so the second attempt and installation succeeds.

`pip install -U urllib3` (to test if an upgraded version helps, current is 1.12) doesn't work since it refuses to override the version provided by the system (which we also can access due to the virtualenv being created with the system side packages flag). I also don't know if that helps and I haven't had dinner yet, so this is how far we are now.

At least we now know a) what causes the problem and b) why does it work the second time you try to install something.

foosel, 2015-11-25:
@WheresWaldo oh, didn't see the comment until doing the above, thanks, that also confirms my suspicion that we have a new pip version here. But as it turns out, that (for once :P) doesn't appear to be the cuprit.

In any case, the problem isn't with OctoPrint, it's with the jessie environment which by default is buggy when it comes to pip-installing packages via https-URL off other places than pypi. YAY *rolls eyes*

foosel, 2015-11-25:
Some further poking reveals [this in the changelog of urllib3](https://pypi.python.org/pypi/urllib3#id7):

> PyOpenSSL: Handle ZeroReturnError exception. (Issue #520)

So yep, we need a new version of urllib3 on this image. Only two options for this really: Use `python-urllib3` package from Debian sid, or create a proper virtualenv without system side packages, install everything (OctoPrint, patched pyserial, urllib3, pyopenssl etc) into that.

Personally I'm voting for the second option because that would also solve a bunch of other problems (e.g. having to globally deinstall `python-serial`) and get away from the horrid mixed scenario (some packages from the system, some from the venv) that we are driving now.

But now: dinne ;)

guysoft, 2015-11-29:
Perhaps the updated raspbian in nightly has the updated urllib3? It seems to have fixes of stuff, but no release notes, so I am not sure how to find out.

foosel, 2015-11-29:
Since no newer version than 1.9.1 is available in jessie: definitely not.

guysoft, 2015-12-05:
So can we close this?

foosel, 2015-12-05:
Don't you want the nightly build to actually succeed first? :P

guysoft, 2015-12-05:
Yes we will wait and see :)

foosel, 2016-03-13:
We forgot to close this, plugin installation has been verified as working in the nightlies repeatedly now I think (by me for example, and probably most of the people using the nightlies now for their rpi3s and not complaining about plugins not working ;))

closed 2016-03-13 by foosel