guysoft/OctoPi #156: [request] OctoPi for Banana Pi M2 / M3

Lord2Vader, 2015-11-05:
Would be a nice thing!

guysoft, 2015-11-05:
@Lord2Vader We have a build of the M1, does the M2 and M3 have a distribution?

Lord2Vader, 2015-11-07:
Due to other cpu it wont work. I tried ;) Doesnt even start :(

guysoft, 2015-11-24:
@Lord2Vader Do you have a bananapi image I can try and build over?

Lord2Vader, 2015-12-10:
Ive got a new power supply and hopefully it works and i will try to make a clean image.

guysoft, 2015-12-16:
@Lord2Vader Any news?
We are sorting out the nightly builds and conformation the bananapi image works would be nice.
Mine broke so I can't test anymore

guysoft, 2015-12-17:
@Lord2Vader Ok we found a bug in the bananapi build, a new nightly is on the way to fix that. see https://github.com/guysoft/OctoPi/commit/8af6ab9f3ab2f0808dbb421c7c98a55443e6bac8

guysoft, 2016-01-06:
@Lord2Vader There was an issue with the bananapi build. It was fixed, and the nightly should use the right image now, could you test that on the bananapi? I have none to test with.

guysoft, 2016-06-04:
See https://github.com/guysoft/OctoPi/issues/232 , we can make all bananapi builds using Armbian.

guysoft, 2016-07-24:
I am closing this issue unless anyone has a problem with that. Since Armbian seems to solve the problem.

guysoft, 2018-01-07:
Closing, information moved to https://github.com/guysoft/CustomPiOS/issues/7

closed 2018-01-07 by guysoft