guysoft/OctoPi #158: Nightly build missing OctoPrint?

beriberikix, 2015-11-19:
I downloaded the [lastest nightly](2015-11-17_2015-09-24-octopi-jessie-0.13.0.zip) and was expecting Octoprint to be configured to autostart. There is a .octoprint folder but I don't think Octoprint is installed anywhere, virtualenv or otherwise. Is that an oversight or known issue?

guysoft, 2015-11-19:
What nightly file did you download?
Please give the file name.

Guy

guysoft, 2015-11-19:
Well spotted,
The last few nights there was a connection problem that was not detected by the build scripts, fixed that and started a new build.
It looks like the connection problem is not present now, and if it ever returns the build would be marked as a failure and not reach the nightly folder.
The build should finish in about an two hours or so:
http://gnet.homelinux.com:8080/job/OctoPi/338/console

beriberikix, 2015-11-20:
:+1: will try the new build tonight.

beriberikix, 2015-11-22:
Verified fix, thanks!

closed 2015-11-22 by beriberikix