guysoft/OctoPi #160: Add support for Waveshare Spotpear TFT screens

WheresWaldo, 2015-11-25:
This could be a documentation issue but it would be nice to support scripts for the Waveshare Spotpear TFT Touchscreen along with the others included in your image. http://www.waveshare.com/product/modules/oleds-lcds/raspberry-pi-lcd.htm

Anything 4" and below are plug SPI devices and they already have both a Wheezy and Jessie image. They also support fbtft.

guysoft, 2015-11-25:
I can't really understand how to install them without access to the hardware, or a step by step command guide. Just linking me to a product does not help.

The guide they have links to a custom kernel from what I can figure (its 64MB), and no explanation beyond that:
http://www.waveshare.com/wiki/4inch_RPi_LCD_%28A%29

WheresWaldo, 2015-11-25:
I understand completely,  I had to do a lot of searching myself to make
sure it worked with OctoPi. Unfortunately I did so many things I can't
remember the actual steps to make it work.  So I guess you can close this
feature request.

guysoft, 2015-11-25:
Well, you could execute `history` in your command line and see all the commands you executed.
You can then find all the commands you used and try and reproduce an install on a fresh OctoPi install, then we can turn those commands in to a script and ship that. Other than that, I cam not a psychic and can't command the developers intended us to run any better than you.

skidrash, 2015-12-23:
This would be great (the waveshare setup was a pain in the neck).  I couldn't figure out how to dl the files so I had to do it off a thumb drive. And I never did get the touch working with octopipanel, so I gave up... If/when I try again I'll try to take notes.

WheresWaldo, 2015-12-23:
Sorry I didn't respond to this earlier, but history is limited and I was
well past the limit when I made this suggestion. If and when I build
another OctoPi box (actually planning this after the New Year), I will try
to take notes. The Waveshare SpotPear is becoming more and more popular as
it is full featured and included complete screen rotation as well.

guysoft, 2018-03-03:
Waveshare screens use a proprietary codebase and are not open source.  We can't ship that. We can only add scripts to install this manually