guysoft/OctoPi #162: VNC to desktop

domwakeling, 2015-11-27:
Is it possible to use VNC to access the 'desktop' when running a RPi headless? I've tried following the guide at https://www.raspberrypi.org/documentation/remote-access/vnc/ but when I try starting the VNC server at either :0 or :1 I get a message that it's already in use by octopi ...

foosel, 2015-11-27:
I would *strongly* advise against trying to run a desktop (be it directly or via VNC) alongside OctoPrint on an RPi, especially when printing. There is no need to fire up a resource hungry X server plus window manager plus whatever else for the functionality that OctoPi intends to provide, and it might actually cause problems when those resources are needed for the actual printer and print server operation.

domwakeling, 2015-11-27:
OK, thanks

closed 2015-11-27 by domwakeling