guysoft/OctoPi #176: OctoPI Panel using OLD RPI display (9bit SPI)?

MasterTom99, 2015-12-15:
I just bought a used RPI display from Watterott to use it as OctoPI-Panel. Iwas not aware that the version of 2014 uses 9bit SPI instead of 8bit. So it is not compatible. From Watterott I found only one image from 2014-06-20 that works with my display. When I start the script to activate the RPI-display, the system does not startup anymore. 
Does anyone know how to use the OLD Watterott 2.8" display with the latest OctoPI Panel? My knowledge of Linux/Raspberry is very basic. Hopefully someone can help me.

guysoft, 2015-12-16:
I don't know if anyone tried.
Watterott have a guide how to set up the 9bit SPI display on any image, thats how we made the scripts shipped with OctoPi, take a look at: https://github.com/watterott/RPi-Display/blob/master/docu/FBTFT-Install.md#rpi-display-9-bit-spi

MasterTom99, 2015-12-16:
I tried to follow the link step by step, but I didn´t succeed. This is what I did:

$ sudo apt-get update
$ sudo rpi-update
$ sudo reboot

$ sudo apt-get install tslib libts-bin
$ sudo apt-get install xinput
$ cd /tmp
$ wget http://tronnes.org/downloads/xinput-calibrator_0.7.5-1_armhf.deb
$ sudo dpkg -i -B xinput-calibrator_0.7.5-1_armhf.deb
$ rm xinput-calibrator_0.7.5-1_armhf.deb

append "dtparam=spi=on" to the /boot/config.txt

$ sudo modprobe fbtft_device name=mi0283qt-9a gpios=reset:23,led:18 speed=32000000 rotate=270
-> the display went to white

append "fbtft_device name=mi0283qt-9a gpios=reset:23,led:18 speed=32000000 rotate=270" to /etc/modules

After reboot the screen is white, but nothing else happens. I guess I made something wrong, but my Linux knowlegge is not enough to make it work. Do you have any ideas what I can try next?

Maybe it´s easier to modify your installation script and use it for installation?

Thomas

MasterTom99, 2015-12-18:
I found a very easy solution for the problem. You can bring the old display to 8bit SPI mode by changing three jumpers. After that it is like the newer version.

https://github.com/watterott/RPi-Display/blob/master/docu/FAQ.md#spi-mode

guysoft, 2019-07-02:
OctoPiPanel is legacy, closing.

closed 2019-07-02 by guysoft