guysoft/OctoPi #178: Watterott RPI display, touch not working

MasterTom99, 2015-12-18:
Sorry to open the 2nd issue within 3 days. I have problems in activating the touch of my Watterott RPI 2.8" display. I did the following:

- Copied V0.12 of the OctoPI and installed it via Win32DiskImager
- Bootet, and connected via Putty.
- cd /home/pi/scripts
- sudo ./enable.rpi-display
- sudo reboot
-> now I could see the console on the RPI display.
- Raspi-Config was loaded automatically, so I expanded the file system and rebootet
- Logging in (pi, raspberry)
- cd /home/pi/scripts
- sudo ./calibrate-rpi-display
- I expected to see the screen with the crosshairs, but it flashes only shortly. On the screen I could read: xres=320, yres=240; selected device is not a touchscreen I understand; Took 1 samples...; and some (virtual) values...

My understanding was, that the image is preconfigured for the RPI display and this are the only steps I have to do. Did I miss something? I know the information on https://github.com/jonaslorander/OctoPiPanel, but I do not know, which of this steps have been done already in this image. Many of the information I found was not up to date. If someone can support me, I can offer to write a step-by-step guide how to activate the panel starting from the standard OctoPI installation. Maybe this can help other beginners in future ;-)

guysoft, 2015-12-19:
Did not have this problem, you can try the other calibration tool that lets you tap the screen, look at xinput_calibrator:
https://github.com/watterott/RPi-Display/blob/master/docu/FBTFT-Install.md#6-calibrate-touchpanel-optional
``sudo DISPLAY=:0 xinput_calibrator --device "ADS7846 Touchscreen" --output-type xinput``

You could ask on the issue tracker of OctoPiPanel, since I never really had this problem.

MasterTom99, 2015-12-19:
In the script "calibrate-rpi-display" you can read:
sudo TSLIB_FBDEVICE=/dev/fb1 TSLIB_TSDEVICE=/dev/input/event0 ts_calibrate

On the Watterott page you can find:
sudo TSLIB_FBDEVICE=/dev/fb1 TSLIB_TSDEVICE=/dev/input/touchscreen ts_calibrate

If I change "event0" in the script to "touchscreen" everything is working fine.

guysoft, 2019-07-02:
OctoPiPanel is legacy, closing this

closed 2019-07-02 by guysoft