guysoft/OctoPi #180: Do I Need to complie the source When I modify some js file?

dilon007, 2015-12-18:
I want to add some code into js file which on /octocprint/src/static/js, and if I want to see the result,I need to complie the source. 
Now I use  the commad "python setup.py install " to complie the source,and restart the octoprint with commad "service octoprint restart".
But  when I open the web page with url "http://octopi.local" I can not see the result of I modify the js.
which the  the above-mentioned step I did is wrong, can you give me help?

guysoft, 2015-12-19:
This should be posted in the OctoPrint issue tracker at: https://github.com/foosel/OctoPrint/issues/
This issue tracker is for OctoPi issues, the operating system, not OctoPrint, the software that runs on it.

guysoft, 2016-05-01:
Closed due to no reply

closed 2016-05-01 by guysoft