guysoft/OctoPi #183: Bananapi M1 doesn't work (nightly+stable)

okercho, 2015-12-29:
Hi!

I've been trying several IMGs from the repo with my bananapi, and I wasn't able to use any of them, as I'm having different issues :(

2015-01-31-octopi-bpi-r1-0.13.0-2015-12-18.img -> Boot up, but no usb-serial drivers for the kernel, so I'm not able to use my printer (lsmod empty)
2015-01-31-octopi-bpi-r1-0.12.0-2015-06-15.img -> Same, boot up, but no usb-serial drivers (lsmod empty)

2015-05-05-octopi-wheezy-0.12.0-2015-07-02.img -> Doesn't boot up
2015-11-21-octopi-jessie-0.13.0-2015-12-17.img -> Doesn't Boot up (only red light, no green light)

Maybe I'm doing something wrong? I have a Rpi too, and I was able to use the stable version provided, and worked perfectly with my printer (Prusa i3 Hephestos).

Thanks!

guysoft, 2015-12-30:
There is a problem with the nightly builds at the moment, you will have to wait I am afraid.

okercho, 2015-12-30:
Ok, no worries, I'm using a Rpi for the moment.

I was looking for the way to change kernel in Raspbian, as the problem with older versions is that usbserial module is not compiled with the shipped kernel... How can I swap kernel? In the lib folder I found another two kernels that has that module compiled, but I don't know how to change kernel in raspbian though.

Regards

guysoft, 2016-01-06:
Hey @okercho 
Can you try the last nightly?
There was a mixup with the images and it should work now:
https://storage.googleapis.com/octoprint/bananapi-m1/2015-01-31-octopi-bpi-r1-0.13.0-2016-01-06.zip

guysoft, 2016-05-01:
Fixed in 0.13.0

closed 2016-05-01 by guysoft