guysoft/OctoPi #184: Timelapse Compilation - GPU and queue?

AcidBurns, 2016-01-01:
Happy new year at first ;)

I got two enhancements for octopi:
- At first, just a suggestion, there is no queue for the timelapse. So I have to wait for the last timelapse to finish rendering before I can start a new print. There should be a queue or something buffering the print, this could be done by simply adding a hour-minute-stamp to the tmp directory where the jpgs are saved (/home/pi/.octoprint/timelapse/tmp0350 instead of simply tmp)

- The second thing is: Why is there no use of the Raspberry GPU for those renderings? A preversion which needs manual filename input can be found in the scripts.zip (or in the text here). If there's a need I can code some bash scripts which accept the same parameters as avconv, so the script can be manually exchanged via the "path to avconv"-variable. 

The comparison for a ~2h print between the ocotpi standard cpu mpg encoding and the gpu enabled h264/mp4 encoding on a raspberry 2 with 4 threads:
CPU: real 43m42.090s, user 71m38.760s, sys 0m43.810s
GPU: real 02m12.324s, user 01m50.230s, sys 0m5.230s

To render the movies, everything is already installed on octopi, just run this command:
gst-launch-1.0 multifilesrc location=/home/pi/.octoprint/timelapse/tmp/tmp_%05d.jpg index=0 caps="image/jpeg,framerate=\(fraction\)24/1" ! jpegdec ! queue ! omxh264enc target-bitrate=15000000 control-rate=variable ! video/x-h264,width=1280,height=720,framerate=24/1,profile=high ! h264parse ! mp4mux faststart=true ! filesink location=/home/pi/.octoprint/timelapse/NAME.mp4

The zip also has different containers (avi, mp4, mkv), where mp4 is recommended. avi doesn't display a thumbnail in windows and mkv has 3 seconds blank at the beginng:
[scripts.zip](https://github.com/guysoft/OctoPi/files/75952/scripts.zip)

AcidBurns, 2016-01-02:
Here's the parameterline compatible script which uses the raspberry pi gpu to render videos.

keep in mind that you have to change the timelapse.py (
/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.8-py2.7.egg/octoprint/timelapse.py) to also display *.mp4 files (change the *.mpg to *.mp*).

[avconv.txt](https://github.com/guysoft/OctoPi/files/76281/avconv.txt)

guysoft, 2016-01-02:
That could be a cool plugin/improvement if we can detet GPU.
@foosel ?

AcidBurns, 2016-01-02:
a note up front: the script will only work on systems where gstreamer with the omx package is installed. I'm not sure if this is possible for other linux systems, especially the embedded ones.. so detecting the gpu doesn't really help if you can't get hold on gstreamer and omxh264enc.

Due to the fact that the avconv.txt script is now parameter compatible to the avconv call from octoprint:
- just put the script into the octoPI (!!) release
- modify the octoPRINT (!!) code to also list mp4, avi and/or mkv files in the timelapse section
- make a tickable box in the webcam section (something like "enable GPU processing (only RPi)").
the box has to set threads to 1 and call the script instead of the /usr/bin/avconv exec.

To circumvent errors one can simply check for the System by running:
sudo lshw -short | grep system | awk '{ print $2 }'
If it puts out "Raspberry" it's running on a raspberry pi and supports the method above. No need to say that this command is present on all debian distros.
Attention: the lshw command needs about 2-3 seconds to process.
rpis can also be indentified by cat /proc/cpuinfo, but that's more complicated.

2bitoperations, 2016-03-05:
@AcidBurns thanks for putting together the avconv.txt wrapper script. I've tested it on my pi2 and it works a treat - much faster encodes and video that looks waaaaaaay better. 

I opened a tiny pull request that lets octoprint see and manage the mp4 files that get created by the wrapper script. Hope we can get it merged to a release eventually. Would you mind if I included your wrapper script too in the pull request?

AcidBurns, 2016-03-05:
sure you can include it. To my mind the stuff here is GPL V3 CC-by-sa if I read it correctly, so my script would also be; that's why I included my name and email in it.
HOWEVER besides the "legal issues": the newest octoPI version (0.13; beta status for now i think) was build on jessie, it no longer has the gstreamer installed. **give me til monday** and I'll rewrite the script to check/install gstreamer and all necessary plugins and stuff to run.

OctopusProteins, 2016-03-18:
@AcidBurns, could you post step by step instructions to doing this? I tried, but all the renders failed. Thanks!

OctopusProteins, 2016-03-18:
@foosel, perhaps you could make all of this part of the next release. (.mp4 exporting, gpu encoding, etc.) I think octoprint would benefit from it.

ntoff, 2016-04-28:
This seems to choke if the images aren't the same size as set in the video/x-h264 parameters. Removing the width and height constraints seems to fix it and render out at the input image size, at least for me anyway. Trying it out on a raspberry pi 3.

OctopusProteins, 2016-06-11:
@2bitoperations, how did you set the threads to 1? Probably missing something very obvious. Also, did you change ALL the .mpg into .mp?

ManuGithubSteam, 2017-09-17:
Hi
What must be done to get this to work ? I would prefer a step to step guide :-)
Thanks!

ManuGithubSteam, 2017-09-17:
Thanks for this!
How can i check if the GPU is really used ?
Thanks!

guysoft, 2017-09-17:
Please don't spam. There is no guide.