guysoft/OctoPi #185: Not launching on Jessie

kd7eir, 2016-01-06:
Anyone having luck with the latest nightlies built on Jessie? I cannot get octoprint to start with any of the Jessie nightlies. If I type sudo service octoprint start I get the following error -

Failed to start octoprint.service: Unit octoprint.service failed to load: No such file or directory.

I tried to update and I got the following error at the end -

Failed to restart octoprint.service: Unit octoprint.service failed to load: No such file or directory.﻿

guysoft, 2016-01-06:
What nightly are you using? Put the date of it

kd7eir, 2016-01-06:
2015-12-19_2015-11-21-octopi-jessie-0.13.0

guysoft, 2016-01-06:
Try the one from today:
2016-01-06_2015-11-21-octopi-jessie-lite-0.13.0.zip
I don't remember the state of OctoPi a month ago.

kd7eir, 2016-01-06:
That worked. Thank you.

closed 2016-01-06 by kd7eir