guysoft/OctoPi #187: OctoPiPanel crashs at startup

Edurobot, 2016-01-06:
I've tried to install OctoPiPanel on a Raspberry Pi B+ with a 2.8'' Adafruit capacitive screen. When I start OctoPiPanel, it crashs a few seconds after startups.

Here is the errors:
```
pi@octopi ~ $ sudo python ~/OctoPiPanel/OctoPiPanel.py
sh: 1: cannot create /sys/class/gpio/gpio252/direction: Directory nonexistent
sh: 1: cannot create /sys/class/gpio/gpio252/value: Directory nonexistent
OctoPiPanel initiated
OctoPiPanel started!
---
Traceback (most recent call last):
  File "/home/pi/OctoPiPanel/OctoPiPanel.py", line 596, in <module>
    opp.Start()
  File "/home/pi/OctoPiPanel/OctoPiPanel.py", line 194, in Start
    self.get_state()
  File "/home/pi/OctoPiPanel/OctoPiPanel.py", line 346, in get_state
    except requests.exceptions.ConnectionError as e:
AttributeError: 'module' object has no attribute 'exceptions'
```

guysoft, 2016-01-07:
Did you run the install script at ``/home/pi/scripts/enable-adafruit-pitft`` ?

XScorpion2, 2016-03-10:
That script corrupts something and prevents my Pi 2 from booting.

markwal, 2016-03-10:
I think you want to look at this guide after you have OctoPi up and running: https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/software-installation

Note the bit at the bottom where it says don't do rpi-update after that because you'll lose the kernel modules you just built.

Also, to have it work with OctoPiPanel, you need to make sure you haven't directed the console to the PiTft because OctoPiPanel use PyGame and it wants the framebuffer all to itself.

markwal, 2016-03-10:
Or use the script on this page (not the image): https://learn.adafruit.com/adafruit-pitft-28-inch-resistive-touchscreen-display-raspberry-pi/easy-install

And answer no to the prompt about wanting the console on the PiTft

XScorpion2, 2016-03-10:
ok, got it back up and running. It was the console issue

guysoft, 2019-07-02:
OctoPiPanel is Legacy, closing.

closed 2019-07-02 by guysoft