guysoft/OctoPi #192: $HAPROXY in /etc/default/haproxy and /etc/init.d/haproxy must match

ariporad, 2016-02-02:
Hi! Thanks for the awesome project!

It would appear that the path to HAProxy must match in `/etc/default/haproxy` and `/etc/init.d/haproxy`, because `/etc/init.d/haproxy` checks for HAProxy existence before loading `/etc/default/haproxy`.

Also, It would be really helpful if there was a comment in `/etc/init.d/haproxy` saying to change the config in `/etc/default/haproxy`.

I couldn't figure out where those files come from, otherwise I would submit a PR.

Thanks!

guysoft, 2017-07-23:
@foosel is this relevant? I am not sure what exactly its about.

Don't ask for PR to PR :) 
I won't pull anything, but if it solves a problem and does not create a new one I will

foosel, 2017-07-24:
AFAIK the init scripts and default files for haproxy are not under OctoPi's control but come from the haproxy Debian package that gets installed. So that should ideally be fixed upstream, not here.

guysoft, 2017-07-24:
You can post a bug report to HAProxy package maintainer at:
https://bugs.debian.org/cgi-bin/pkgreport.cgi?pkg=haproxy;dist=jessie

@kaplanlior Any idea who we can report this to?

guysoft, 2018-01-07:
Is this still relevant? Closing if no reply

closed 2019-07-02 by guysoft