guysoft/OctoPi #193: `sfdisk: bad input` when building with raspbian jessie

jminardi, 2016-02-03:
I was trying to build an octopi image based off raspbian jessie. I used the following custom config.local:

```
OCTOPI_OVERRIDE_HOSTNAME=myname
OCTOPI_INCLUDE_OCTOPIPANEL=no
OCTOPI_INCLUDE_CURAENGINE=no
```

Here is the last few lines of the error message

```
e2fsck 1.42.9 (4-Feb-2014)
Pass 1: Checking inodes, blocks, and sizes
Pass 2: Checking directory structure
Pass 3: Checking directory connectivity
Pass 4: Checking reference counts
Pass 5: Checking group summary information
/dev/loop0: 92745/236176 files (0.1% non-contiguous), 534999/944128 blocks
+++ e2ftarget_bytes=2415919104
+++ e2ftarget_blocks=4718593
+++ echo 'Resizing file system to 4718593 blocks...'
Resizing file system to 4718593 blocks...
+++ resize2fs /dev/loop0 4718593s
resize2fs 1.42.9 (4-Feb-2014)
Resizing the filesystem on /dev/loop0 to 589824 (4k) blocks.
The filesystem on /dev/loop0 is now 589824 blocks long.

+++ losetup -d /dev/loop0
+++ trap - EXIT
+++ new_end=4849665
+++ echo 'Resizing partition to end at 131072 + 4718593 = 4849665 blocks...'
Resizing partition to end at 131072 + 4718593 = 4849665 blocks...
+++ echo 131072,4849665
+++ sfdisk -N2 2015-11-21-raspbian-jessie.img
Warning: 2015-11-21-raspbian-jessie.img is not a block device
Disk 2015-11-21-raspbian-jessie.img: cannot get geometry

Disk 2015-11-21-raspbian-jessie.img: 478 cylinders, 255 heads, 63 sectors/track
Old situation:
Units = cylinders of 8225280 bytes, blocks of 1024 bytes, counting from 0

   Device Boot Start     End   #cyls    #blocks   Id  System
2015-11-21-raspbian-jessie.img1          0+      8-      8-     61440    c  W95 FAT32 (LBA)
2015-11-21-raspbian-jessie.img2          8+    478-    471-   3776512   83  Linux
2015-11-21-raspbian-jessie.img3          0       -       0          0    0  Empty
2015-11-21-raspbian-jessie.img4          0       -       0          0    0  Empty
Warning: given size (4849665) exceeds max allowable size (0)

sfdisk: bad input
```

jminardi, 2016-02-03:
I was trying to build this from Ubuntu 14.04 running in a parallels VM

foosel, 2016-02-03:
Try if reverting commits 3aac0c8783b3fd90455cb9511178f3cb93fe248d and 1fdff7b6cbac74459005e4b5b6b3f0a3b1aa7a3c helps you. Helped me.

jminardi, 2016-02-05:
That seemed to fix it. Should those commits be reverted on master?

guysoft, 2016-03-06:
Revert is now in devel branch

closed 2016-03-06 by guysoft