guysoft/OctoPi #196: How would one compile OctoPi for x86?

KeiroD, 2016-02-08:
Hi!

I was curious as to whether or not it'd be possible to compile `OctoPi` on `x86/x86_64` hardware rather than the `arm` hardware as is currently the case?

Use-case: I'd like to make use of OctoPi as a pre-built image to install on a mini-ITX build that should run OctoPi/OctoPrint rather nicely.

I've noticed that here and there, my `Raspberry Pi` seems to struggle with running OctoPi/OctoPrint, which is why I was considering going this route. My problem though is, I'm not sure if it'd be feasible for me to do so.

As far as I can tel, it seems to only require a kernel change to one that works on the x86/x86_64 hardware?

guysoft, 2016-02-08:
The architecture is not an issue, the question is if you have an image for you device in question.
I am looking in to getting the hummingboard to work, that uses a single partition, if that works you have a good chance of porting it to some debian-based x86 image.

guysoft, 2018-01-04:
Old, closing

closed 2018-01-04 by guysoft