guysoft/OctoPi #197: Print stutters when loading web interface

thedoble, 2016-02-11:
Hi folks

I have found that when loading the web interface, my OctoPi pauses or stutters during the print for a second or two. The steppers stop moving briefly while the web interface loads. This is exacerbated if I load the interface in two browsers are once.

Obviously this is bad because it can cause blobs in the print.

I have tried raising the priority of the Octoprint process, but this doesn't seem to have had much effect.

Any tips on how I can get rid of these stutters?
My printer is a Rostock Max V2 connected at 250000 baud.

Thanks

foosel, 2016-02-11:
Which OctoPrint version is this? If not 1.2.9 or at least 1.2.8, try updating please. There were a lot of changes with regards to aggressive server side caching to prevent stuff like this from happening (of course, that requires that you don't perform shift-reloads in your browser, which would force the full page to be re-rendered, which is a quite cpu intense process).

The part of OctoPrint that serves the page and the part that talks to your printer is the same process, so raising the process priority will not achieve anything here.

Also, this is probably better placed in the [OctoPrint bugtracker](https://goo.gl/GzkGv9).

thedoble, 2016-02-14:
Thanks for your reply.
I am running Octoprint 1.2.9.
I suspect that if I post in the octoprint bugtracker, they will refer me back here as the problem is likely due to the CPU constraints associated with the Raspi.
If I disable my webcam, the printer doesn't stutter when loading. I suspect the load from webcam + octoprint is what causes the stuttering.

guysoft, 2016-02-14:
I found this happens when I use the touch UI plugin.

thedoble, 2016-02-24:
I note that in the terminal, when the printer freezes, it responds 'Wait' and the commands are lost.

Opened an issue on Octoprint page: https://github.com/foosel/OctoPrint/issues/1241

foosel, 2017-02-28:
Closing since this got moved to OctoPrint's own tracker.

closed 2017-02-28 by foosel