guysoft/OctoPi #198: OctoPrint: 1.2.9 Locking up Pi after print finishes, Timelapse finishes, present upon reboot

crprintoctopi25, 2016-02-12:
Hi, My octoprint setup works well!

Problem is, it seems to be locked up once a print is done.  I have to pull power and power back up.  I have re-imaged at least 4 times (on 2 different cards).

What would be the next step for troubleshooting?  When "locked up" the red light is on solid, and the wipi light is not on at all as it usually is.  The green light doesn't do anything at all.

When working all seems fine!  Installation went exactly as described in the directions without a hitch!  Auto upgraded to current v. 1.2.9.  Re-Booted fine once installed, etc.

I also have encountered the printer slow to a complete halt when using the Web Interface re: #197 

**I have the following all brand new this week:**

**Micro SD Card**
https://www.amazon.com/gp/product/B00RVNGGT8/ref=oh_aui_detailpage_o00_s01?ie=UTF8&psc=1

**Pi 2 Model B**
https://www.amazon.com/gp/product/B00T2U7R7I/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

**Wi-Pi Adapter**
https://www.amazon.com/gp/product/B00BDW6D7I/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

**Power Supply**
https://www.amazon.com/gp/product/B00VH8ENXE/ref=oh_aui_detailpage_o00_s00?ie=UTF8&psc=1

**USB Cable for Power - using a super nice one that came with BOSE Speaker.
Ordered these cables, but not in till next Wed:**
https://www.amazon.com/gp/product/B00MLSY6MM/ref=oh_aui_detailpage_o01_s00?ie=UTF8&psc=1

**Webcam:**
I already had - works well - took over 4 timelapse vids of prints already
Logitech USB - forgot model - but was on list of compatible cameras

guysoft, 2016-02-12:
What version of OctoPi?

markwal, 2016-02-12:
My first suspect would be wi-fi.  It may not be your card, but any of the little nano adapters are fairly sensitive to their location, but also some of the wi-fi adapter driver's are gimpy (IMHO).  If you are able, I'd test with a hard wire ethernet cable into your router and see if it happens again.  If that's not possible, you may want to set up a ping from the pi (via ssh) to your computer and see how stable it is.  You could also check dmesg.

crprintoctopi25, 2016-02-12:
Hi, Thank you for your comments:

I truly appreciate your help!  Just Fyi, I ran nothing on the Pi last night since hard restarting around say 8 pm last night.  So it's been up and fine still for about 9hrs.  I also installed VNC and did the apt-get update & upgrade thru the terminal and once completed, re-booted and was able to get into via SSH & VNC after successful reboot.  The only thing I didn't do surprisingly, was run a print last night :-) which at least let's us know it seems the pi is semi-stable....I guess?  I left it with the VNC session open and SSH and both are still live right now.

**guysoft,**

OctoPi Ver:
2015-07-02_2015-05-05-octopi-wheezy-0.12.0.zip / Size: 1,105,101
---->  2015-05-05-octopi-wheezy-0.12.0.img / Size: 2,200,000

Downloaded from: 
https://github.com/guysoft/OctoPi  ----> https://octopi.octoprint.org/latest
[image]

**markwal,**

1. Unfortunately, at the moment, I'd have to drill / run across a door opening for ethernet - but is on my to do's somehow regardless someday.  For now, the router is on same floor - about 15 - 18' away.  It's a: D-LINK Product: DIR-655, Firmware Version: 1.21, Hardware Version: A3 [looks like they have a 1.37 firmware now from 6/11/13]
2. How do you recommend I setup a ping from the PI? Would that then save a log to refer to?
3. I'm not familiar with dmesg either.  I checked out the basic command, snytax and options.  Again, if you could be so kind as to let me know what you'd do to utilize this command.

Thank you so much again in advance!  I am really impressed with this little guy and it's capabilities!!!  Far cry from the old days of bnc network and loud seagate drives :-)

guysoft, 2018-01-07:
Looks fixed, closing if no reply

closed 2019-07-02 by guysoft