guysoft/OctoPi #199: SDL library in Jessie not compatible with pygame (so pitft won't work).

fcawth, 2016-02-13:
Using a recent nightly build (because I wanted to use Jessie on the Pi2), I had everything set up and the adafruit 2.8 capacitive touchscreen showing pitft, but the touchscreen was not working.  The touchscreen worked fine in all the test applications.  It turns out to be an issue with SDL 2.x and pygame.  If I would have made it to the end of the tutorial, I would have found this on adafruit's site: 
[https://learn.adafruit.com/adafruit-2-8-pitft-capacitive-touch/pitft-pygame-tips](https://learn.adafruit.com/adafruit-2-8-pitft-capacitive-touch/pitft-pygame-tips).  SDL should probably be downgraded in the development branch of octopi.  Works fine with this change.

guysoft, 2016-02-22:
@jonaslorander 
Might be a good idea to include:

```
    #!/bin/bash
      
    #enable wheezy package sources
    echo "deb http://archive.raspbian.org/raspbian wheezy main
    " > /etc/apt/sources.list.d/wheezy.list
     
    #set stable as default package source (currently jessie)
    echo "APT::Default-release \"stable\";
    " > /etc/apt/apt.conf.d/10defaultRelease
     
    #set the priority for libsdl from wheezy higher then the jessie package
    echo "Package: libsdl1.2debian
    Pin: release n=jessie
    Pin-Priority: -10
    Package: libsdl1.2debian
    Pin: release n=wheezy
    Pin-Priority: 900
    " > /etc/apt/preferences.d/libsdl
     
    #install
    apt-get update
    apt-get -y --force-yes install libsdl1.2debian/wheezy```

guysoft, 2019-07-02:
OctoPiPanel is legacy, closing.

closed 2019-07-02 by guysoft