guysoft/OctoPi #201: Webcam jitter

kentmartin99, 2016-02-16:
Hi, I am using octopi on a raspberry pi, and the bottom half of the webcam picture is being offset randomly. I am using a microsoft hd-3000 usb camera. 
I added "camera_raspi_options="-i "./input_uvc.so -d /dev/video0 -r 1280x720 -f 20" -o "./output_http.so -p 8080" into the octopi.txt file in the boot folder. Was supposed to be for this camera, made no difference to the problem.
I am using version 1.2.2 of octoprint.
How can I correct thi problem?
Thanks.
[image]

foosel, 2016-02-16:
The webcam image is provided by mjpg-streamer - my guess is you get the same distortion when going to the webcam server itself at http://octopi.local:8080/, OctoPrint/OctoPi doesn't manipulate the image in any way. Can you verify this?

If so, this is probably an issue between mjpg-streamer (which is a project on its own, OctoPi merely bundles it) and your camera. It looks like it is hiccuping while retrieving the data from your webcam. I'd suggest experimenting with different resolutions and framerates, maybe some combination solves the issue.

Usually I'd suggest seeking help in the mjpg-streamer bug tracker, but the official SF.net project looks pretty much abandoned. So instead I suggest to post on the [OctoPrint mailinglist](https://groups.google.com/group/octoprint) or the [G+ community](https://plus.google.com/communities/102771308349328485741) instead - maybe someone else is running the exact same camera and can provide you with some insight on this issue.

Alternatively, it might also be that something "happens to the stream" while traveling from the Pi to your browser. I would be surprised by this, but better also try a wired connection instead of a wireless one to make sure it isn't some odd traffic issue (if you are currently using WiFi, if not ignore this).

kentmartin99, 2016-02-16:
Yes the problem occurs of you use the camera streaming only page as well. I'm not sure how to change the frame rates or other parameters? I'll try the mailing list. The camera is supposed to be supported. I am running wifi. I can test with LAN , but I would think this was the issues, seems to update quickly enough the parts that aren't mangled.

foosel, 2016-02-16:
For people finding this via some search or another, further discussion here:

  * [Mailing list thread](https://groups.google.com/d/msgid/octoprint/125041e9-784d-48ac-975d-740100d046f6%40googlegroups.com?utm_medium=email&utm_source=footer)
  * [G+ Community thread](https://plus.google.com/b/110130855001142142895/104530472148180255980/posts/azAViesbBDP?cfem=1)

guysoft, 2016-05-01:
People reported using 0.13.0 removes camera jitters, could you test and report if it solves it for you?