guysoft/OctoPi #202: Static IP on wired Ethernet..

tobiasbischoff, 2016-02-17:
Hi, 

somehow I cannot desable dhcp and set  a static IP for the fixed LAN. Is /etc/network/interfaces ignored somehow?

guysoft, 2016-02-18:
What version of OctoPi are you using?
Can you follow the boot sequence on a screen and see if it tries to use
DHCP to get an address?

waynemichel, 2016-02-19:
You now need to edit /etc/dhcpcd.conf and ignore /etc/network/interfaces.

howto: http://www.circuitbasics.com/how-to-set-up-a-static-ip-on-the-raspberry-pi/

foosel, 2016-02-19:
And in case you are wondering why: https://www.raspberrypi.org/blog/another-raspbian-desktop-user-interface-update/

Personally, I'm not a big fan of `dhcpcd`, `/etc/network/interfaces` works just fine, especially headless, but removing that stuff by default would confuse a lot of people probable ("Why doesn't X work like in this Raspberry Pi tutorial over here?!")

jrpmedia, 2016-07-21:
I cannot see _/etc/network/interfaces_ on the USB card when it is in my Mac OSX.
Even if I _show invisible_ files.
Is there another way, on Mac, to access this file location on the USB card?

guysoft, 2017-07-23:
See https://github.com/guysoft/OctoPi/issues/219 , there are instructions how to set it up today with Jessie.

guysoft, 2018-05-11:
Any news on this? Is it working?

guysoft, 2019-04-23:
Closing due to no reply

closed 2019-04-23 by guysoft

IT-Mase, 2024-05-29:
Hi !
I had the same issue ...
Found an answer, for last version which is under raspbian bookworm [here](https://www.abelectronics.co.uk/kb/article/31/set-a-static-ip-address-on-raspberry-pi-os-bookworm) !
It worked like a charm for me !

Ciao !

guysoft, 2024-06-02:
Thanks @IT-Mase if you want you can include it in the wiki under network:
https://github.com/guysoft/OctoPi/wiki/FAQ

SlyLaM, 2024-06-03:
I don't know how it has to be done but I tried.
you can check here:
https://github.com/guysoft/OctoPi/wiki/Set-Static-IP-on-wired-connection

Thanks

guysoft, 2024-06-03:
@SlyLaM good enough. Thanks!