guysoft/OctoPi #203: Serial connection issue on first attempt

jcernato, 2016-02-18:
Hi there, I have my printer connected via internal Serial-port (ttyAMA0) to the Serialport-3 on the Arduino Mega.
Connecting from Octoprint the Terminal-communication states: 
`Changing monitoring state from 'Opening serial port' to 'Connecting'`
`Send: M110`
`Recv: ok`
and after a few seconds
`Changing monitoring state from 'Connecting' to 'Closed'`
`Connection closed, closing down monitor`

Connecting a second time works perfectly fine. Now I'm wondering why this happens, is this an issue of my Marlin-Firmware or could that be a Problem within the octoprint environment? 
It's not that big deal, it's just annoying having to connect twice...

markwal, 2016-02-19:
It does seem like your printers firmware is closing the serial connection.  You might want to post this query to the google group for OctoPrint and see if other people have seen this with Marlin.  It's not really an issue with the octopi image though.

https://groups.google.com/forum/#!forum/octoprint

foosel, 2017-02-28:
As @markwal pointed out, the mailing list as linked above is the better place for these kinds of support requests

closed 2017-02-28 by foosel