guysoft/OctoPi #204: Octopi crashing when trying to update

I-am-me, 2016-02-19:
I recently installed OctoPi 0.12 on my raspberry pi, it comes with OctoPrint 1.2.2.

When I try to update to OctoPrint 1.2.9 it says there was an error.
After I reboot and try to access the OctoPrint UI I get a 503 error saying "Service Unavailable"

I am still able to access the raspberry pi through SSH, I tried restarting OctoPrint from the pi but it still won't work.

markwal, 2016-02-19:
What does ~/.octoprint/logs/octoprint.log say?

I-am-me, 2016-02-19:
I forgot to copy the logs before re-rewriting my SD card. 
I just tried to see if it would crash and it worked, so I guess it was a problem on my end. If it happens again I'll let you know.

Thanks anyway

closed 2016-02-19 by I-am-me