guysoft/OctoPi #205: Weird temperature readings & LCD flashing

I-am-me, 2016-02-22:
Hi,

I am using octopi 0.12 with octoprint 1.2.9 on a raspberry pi B (the first one).

When I plug my printer, a prusa clone called Prusa Xi3, the LCD is flashing weirdly, it displays what is should but flashes.

On the LCD the temperature reading is fluctuating between 22°C and -8°C and on the octoprint UI the temperature graph is not updating every 5 seconds like it should do.
If I send M105 command it will update but will not on his own.

I think it might be a power issue, the raspberry pi not supplying enough power to the printer. 
I tried to use a 1.2A power supply but no change occurred.

guysoft, 2016-02-22:
Can you post a video or something of the flashes, I had something similar, might not be OctoPi related, but the power between the Pi and the printer.

nophead, 2016-02-22:
The RPi should not be powering the printer. The printer should have a 12V
supply and derive 5V from that.

I-am-me, 2016-02-22:
I tried to record the flashes but the camera can't see them. But I can describe : the screen is really dark and lights inconsistently.

My printer need to be connected to 5V (from PC or phone charger) for the arduino to power up. It does not work with the power supply.

markwal, 2016-02-23:
I think you've already figured out the problem, your Arduino isn't getting enough power. I don't think there's any change to OctoPrint or octopi that can help you with that.

I-am-me, 2016-02-23:
Could the lack of power be causing the weird temperature readings ?

nophead, 2016-02-23:
Yes. The temperature is measured by the Arduino. If the power to that is out of spec it might give erroneous results. You want it powered from a well regulated 5V supply, not the random voltage you get at end of a USB cable. That is only really suitable for powering 3.3V devices.

closed 2016-02-23 by I-am-me