guysoft/OctoPi #207: Octoprint requires Raspbian Jessie based distro to work with Raspberry Pi 3

alex-gibson, 2016-02-29:
I have been testing ahead of the launch today, and noticed that the standard OctoPi distro gives rainbow screen of death on the Pi 3, and I could not follow the instructions (due to errors) to install for Jessie, using latest NOOBS distro as at last night.

For hassle free OctoPrint on the Pi 3 I recommend starting with the latest Jessie nightly build, which worked well enough that I demo'd it at the Pi 3 launch party today :)

Happy to contribute testing to get a stable Jessie/Pi3 build!!!

taxilian, 2016-03-05:
I just ran into this as well; I appreciate knowing that I'm not the only one, and I'll proceed following those instructions =]

nophead, 2016-03-05:
I have OctoPrint running on a RPi3 by installing latest NOOBS and then following the OctoPrint install instructions for Linux. Tedious but they work and they also work on Ubuntu on an Orange Pi.

gammy, 2016-03-05:
I installed Raspbian Jessie Lite (2016-02-26), but did so manually rather than via NOOBS. I then followed the OctoPrint installation instructions, except that I checked out v1.2.9(the latest stable version as of writing) rather than using the master. Tedious indeed @nophead! It all works though, including optional dependencies such as mjpg-streamer and cura-engine.

nophead, 2016-03-06:
What it needs is an OctoPrint install script that runs on any Linux and does what the Linux instructions specify. Then you don't really need OctoPi.

guysoft, 2016-03-06:
Hey guys, have you tried the nightly build?
http://docstech.net/OctoPiMirror/nightly/
eg.
https://storage.googleapis.com/octoprint/2016-03-02_2016-02-26-octopi-jessie-lite-0.13.0.zip

pmbdk, 2016-03-06:
The latest nightly build works fine with RPi3 (although I havn't actually tried printing with it yet)

A-Nesh, 2016-03-06:
I have also installed a nightly build (2016-03-02_2016-02-26-octopi-jessie-lite-0.13.0.zip) in RPi 3. The RPi freezes during a print.

seefood, 2016-03-06:
Wow, people got their RP3 quickly :-)
Now @guysoft only needs to take care of a Pine64 version and.... Hey, isn't it time to just create a PPA? :)

foosel, 2016-03-06:
> The Pi freezes half way through a print.

I've run into a similar issue with the Pi3 I just got, mine just sat idly though and suddenly stopped responding altogether, several times. On first look it seems to be power related, switched to a different PSU, haven't been able to reproduce the issue since then (still observing it though). 

Since the Pi3 [does consume more power than the Pi2](https://www.element14.com/community/community/raspberry-pi/blog/2016/02/29/the-most-comprehensive-raspberry-pi-comparison-benchmark-ever), it might be that one simply can't get away with the same power supplies anymore as was still possible with Pi1/Pi2. Just an idea though. I certainly never had a Pi2 just freeze on me with that PSU while doing basically nothing.

A-Nesh, 2016-03-06:
I don't believe that it's a power issue in my case as mine PPi is connected to a ATX PSU which can deliver 25A on a 5V rail.

foosel, 2016-03-06:
After it just happened again, I have to agree, doesn't seem to be power related.

I also have to correct myself, it apparently doesn't freeze completely after all, it just looses wifi connectivity. Funny thing is, it believes it still does have connectivity, has an IP and everything, but cannot be pinged or SSH'd into. Until I ping the router from the Pi3, then suddenly connectivity is back:

```
$ ping -t 192.168.1.204

Pinging 192.168.1.204 with 32 bytes of data:
Reply from 192.168.1.3: Destination host unreachable.
Reply from 192.168.1.3: Destination host unreachable.
Reply from 192.168.1.3: Destination host unreachable.
Reply from 192.168.1.3: Destination host unreachable.
Reply from 192.168.1.204: bytes=32 time=2030ms TTL=64
```

Happens very reproducibly after some idle time. Maybe some power safe feature of the wifi chipset that's misbehaving (like it also happens with some dongles out there)?

@A-Nesh did it really freeze for you during printing (print stopped) or could it be that just the network connection broke down (print didn't stop, but anything relying on network didn't work anymore, so OctoPrint, SSH, ...).

A-Nesh, 2016-03-06:
@foosel my printer just stopped printing, after some printing (an hour or so). The extruder and the heated bed was still on. It happened two times today. I don't know if the whole Pi freeze or not but it was not accessible either by HTTP or SSH. I'll check it next time it misbehave.

foosel, 2016-03-06:
If it even stopped printing, that sounds like a different issue then. We are running the same image, so if it's an image issue, I should hopefully see it too during my tests.

From what you describe it indeed sounds like the whole Pi crashes completely. Do you have a chance to attach a monitor to it? That might show what's happening when it goes down. Also of interest would be your `/var/log/syslog` and you `dmesg`, maybe something got logged there when it went down.

A-Nesh, 2016-03-06:
i have now encountered same issue as you @foosel. It did not stop printing but the network died. I'll connect a monitor next time it happens.

Here are my syslog and dmesg:
[dmesg.txt](https://github.com/guysoft/OctoPi/files/160374/dmesg.txt)
[syslog.txt](https://github.com/guysoft/OctoPi/files/160376/syslog.txt)

A-Nesh, 2016-03-06:
It died again now, after 77%

PING 192.178.1.88 (192.178.1.88): 56 data bytes
Request timeout for icmp_seq 0
Request timeout for icmp_seq 1
Request timeout for icmp_seq 2
Request timeout for icmp_seq 3
Request timeout for icmp_seq 4

A-Nesh, 2016-03-06:
I have connected a monitor to the RPi, there is no signal = the RPi is dead

foosel, 2016-03-06:
[I asked in the G+ community](https://plus.google.com/u/0/+GinaH%C3%A4u%C3%9Fge/posts/g8jUYd8MSWY) if people were seeing this too with OctoPi and the RPi3. So far no definitive answer, but a good question:

> If they're running an ATX PSU have they got any dummy load resistors to flatten out the power? I too thought it'd be no problem but noticed under load my ATX was dropping to 4 volts. Added a dummy load and it stabilized the output. 

and some good advice I have to agree with:

> Check your power supply.  Try it without the printer turned on and see if stays up.  Try with a separate 5V power supply and running a print.  More troubleshooting is needed.
>
> Try a different memory card?﻿

Regarding the wifi going down, that's definitely caused by power management. I disabled it via `/boot/octopi-network` by adding `wireless-power off` to my `wlan0` config, right after the SSID/PSK combo ([basically as described here](http://www.modmypi.com/blog/disable-wifi-power-management)) and since doing that I couldn't reproduce the "un-ping-able" issue any more. Removed the line again and bam, issue returned. So something appears to be broken with the power management of the Pi3's own wifi.

I've seen older RPis freezing (exact same symptoms you are observing) due to issues with the wired interface, power supply issues (especially when under load), USB devices drawing too much power (which is why I prefer to have a powered hub between my Pi's and their peripherals), etc etc.

A-Nesh, 2016-03-06:
ok, thanx. Ill try a separate power adapter and report later

nophead, 2016-03-06:
If you are using an ATX PSU it is much better to use the 5V standby power
rather than the main 5V. It is actually a separate power supply so isn't
affected by the loads on the other outputs and you can also then use the
RPi to control the other half of the PSU.

A-Nesh, 2016-03-06:
Have tried both 5v standby and ordinary 5v rails with no luck. I also believe that it is a power issue. I have  connected usb power monitor and can see that there almost no current consumption when the printer is on. Which means that the power must come from the printer to RPi via USB. I use geeetech gt2650 board. It was never issue with OctoPi on mi Raspberry Pi 1.

nophead, 2016-03-06:
Power should only flow from host to device on a USB cable. If device power is powering the host it is faulty. It wouldn't surprise me from Geeetech.

A-Nesh, 2016-03-07:
I have left the Pi over night connected to a 3A USB power supply and it was dead this morning. I have connected a monitor and as it was not responding i have restarted the Pi. A couple minutes later, i have found the issue. I am new to linux but it looks to me like file system error. 
[image]

foosel, 2016-03-07:
That looks like a kernel panic. Download /var/log/syslog (eg via Womack), upload it somewhere (Dropbox or similar), share the link here. 

The full log output of the kernel panic might tell us what's up, if it's a hardware issue, a flash gone wrong etc.

Just for the record, my pi3 has been up now since my last post yesterday without issues, using the same image and the above mentioned change to disable the Wi-Fi power management.

taxilian, 2016-03-07:
Just as a datapoint, I installed using the raspbian image and then followed the octoprint install instructions and have been printing back to back jobs since Friday night with no issues.

Have not used Wi-Fi our changed any other settings.

A-Nesh, 2016-03-07:
@foosel here is my syslog
[syslog.txt](https://github.com/guysoft/OctoPi/files/160947/syslog.txt)

A-Nesh, 2016-03-07:
@taxilian actually it worked fine for me to when i installed latest Jessie and manually installed OctoPi. I don't know why did i install a nightly build instead when everything worked fine.

foosel, 2016-03-07:
@A-Nesh because giving us early feedback on how the nightlies perform is the only way to root out issues *before* release, and you knew that and wanted to help? ;)

A-Nesh, 2016-03-07:
@foosel yes, it's was pretty much my plan :)

foosel, 2016-03-07:
@A-Nesh I took a look into your syslog. Seems like the issue you made a picture of above didn't make it into the log :/ But there are a LOT of errors with the wifi interface, including some traces while trying to get an IP via dhcp. Have you tried disabling WiFi power management as described above? It's just a complete shot in the dark... if that doesn't help, I'd suggest reflashing/another SD card, to rule out issues with the flash and the card.

Can't help myself, I don't have the feeling that their drivers are all "there" yet, the raspberry forums are also quite full with "doesn't work" posts in the wifi thread at the moment.

Might also be interesting to see if they maybe did things differently between the lite and the full raspbian image. OctoPi uses the lite image.

A-Nesh, 2016-03-07:
@foosel Yes, i also notice a DHCP attempt today. It still connects to the manual IP (.88). My DHCP range is .100 an up. I also took a look in the syslog and couldn't find the error message. Maybe it died died before it could write the log. 

I have used the full Jessie image earlier. I will try to to flash a new nightly image to a new SD card, as soon i find it.  :) just to eliminate sd card issue

I didn't have a chance yet to disable wifi power management, i'll do it first now.

A-Nesh, 2016-03-07:
I have tested fresh nightly image with wireless-power off, on a new SD card.... same issue. Here is my syslog.
[syslog.txt](https://github.com/guysoft/OctoPi/files/161264/syslog.txt)

I am now installing full Jessie image...

A-Nesh, 2016-03-07:
Nope, same freeze during print issue. This time 54%. I give up, back to my RPi1

foosel, 2016-03-07:
Oh, but that syslog was way more interesting:

```
Mar  7 11:56:44 octopi kernel: [ 1483.101088] Unable to handle kernel NULL pointer dereference at virtual address 00000000
Mar  7 11:56:44 octopi kernel: [ 1483.117168] pgd = 80004000
Mar  7 11:56:45 octopi kernel: [ 1483.123041] [00000000] *pgd=00000000
Mar  7 11:56:45 octopi kernel: [ 1483.130335] Internal error: Oops: 17 [#1] SMP ARM
Mar  7 11:56:45 octopi kernel: [ 1483.139454] Modules linked in: bnep hci_uart btbcm bluetooth brcmfmac brcmutil cfg80211 ftdi_sio rfkill joydev evdev usbserial snd_bcm2835 snd_pcm bcm2835_gpiomem snd_timer bcm2835_rng snd uio_pdrv_genirq uio ipv6
Mar  7 11:56:45 octopi kernel: [ 1483.175090] CPU: 1 PID: 42 Comm: kworker/1:1 Not tainted 4.1.18-v7+ #846
Mar  7 11:56:45 octopi kernel: [ 1483.187596] Hardware name: BCM2709
Mar  7 11:56:45 octopi kernel: [ 1483.194773] Workqueue: events brcmf_fweh_event_worker [brcmfmac]
Mar  7 11:56:45 octopi kernel: [ 1483.206209] task: b5b68000 ti: b5b2a000 task.ti: b5b2a000
Mar  7 11:56:45 octopi kernel: [ 1483.216669] PC is at cfg80211_get_bss_channel+0xf8/0x190 [cfg80211]
Mar  7 11:56:45 octopi kernel: [ 1483.228505] LR is at 0xfffffff6
Mar  7 11:56:45 octopi kernel: [ 1483.235206] pc : [<7f12eb48>]    lr : [<fffffff6>]    psr: 60000013
Mar  7 11:56:45 octopi kernel: [ 1483.235206] sp : b5b2bcc8  ip : 00000003  fp : b5b2bcec
Mar  7 11:56:45 octopi kernel: [ 1483.257060] r10: 00000064  r9 : 00000000  r8 : 00000000
Mar  7 11:56:45 octopi kernel: [ 1483.267106] r7 : 000000c5  r6 : 000000b4  r5 : b2d60160  r4 : b3f90095
Mar  7 11:56:45 octopi kernel: [ 1483.279271] r3 : 00000000  r2 : 000000c5  r1 : b3f90084  r0 : 0000000d
Mar  7 11:56:45 octopi kernel: [ 1483.291417] Flags: nZCv  IRQs on  FIQs on  Mode SVC_32  ISA ARM  Segment kernel
Mar  7 11:56:45 octopi kernel: [ 1483.304842] Control: 10c5383d  Table: 32da806a  DAC: 00000015
Mar  7 11:56:45 octopi kernel: [ 1483.315699] Process kworker/1:1 (pid: 42, stack limit = 0xb5b2a210)
Mar  7 11:56:45 octopi kernel: [ 1483.327372] Stack: (0xb5b2bcc8 to 0xb5b2c000)
Mar  7 11:56:45 octopi kernel: [ 1483.335969] bcc0:                   00000000 b2d60160 000000c5 000000c5 00000000 00000000
Mar  7 11:56:45 octopi kernel: [ 1483.352402] bce0: b5b2bd74 b5b2bcf0 7f12fa18 7f12ea5c 80819cdc 00000c11 00000000 00000000
Mar  7 11:56:45 octopi kernel: [ 1483.368889] bd00: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mar  7 11:56:45 octopi kernel: [ 1483.385386] bd20: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mar  7 11:56:45 octopi kernel: [ 1483.401948] bd40: 00000000 00000000 b5247c40 b2d60160 000000d0 000000c5 00000064 00000c11
Mar  7 11:56:45 octopi kernel: [ 1483.418617] bd60: 00000000 b3f90004 b5b2bde4 b5b2bd78 7f251700 7f12f9b4 b3f9000c b2d60160
Mar  7 11:56:45 octopi kernel: [ 1483.435392] bd80: 00000000 00000000 00000c11 00000064 b3f90084 000000c5 ffffeae8 000000d0
Mar  7 11:56:45 octopi kernel: [ 1483.452310] bda0: 8056be54 00000c11 00000800 7f2598a4 b5b2bde4 b5b2bdc0 7f2598a4 b3f90004
Mar  7 11:56:45 octopi kernel: [ 1483.469432] bdc0: b3f90000 b5247c40 00000000 b2d60160 b2e3b248 b2e3b000 b5b2be0c b5b2bde8
Mar  7 11:56:45 octopi kernel: [ 1483.486644] bde0: 7f252204 7f25163c b5b2be88 b2d60340 b5b2bea0 b2d60340 b5247800 b5247c40
Mar  7 11:56:45 octopi kernel: [ 1483.503926] be00: b5b2be5c b5b2be10 7f252474 7f25219c 80056f60 80056a28 ffffffff 00ffffff
Mar  7 11:56:45 octopi kernel: [ 1483.521267] be20: 00000001 b6fc0ac0 0001ce82 00000001 b6fc0ad0 00000013 b2c36098 b2c360a8
Mar  7 11:56:45 octopi kernel: [ 1483.532236] be40: 00200200 00100100 b2c360ac b2c34000 b5b2be74 b5b2be60 7f259de0 7f2523c8
Mar  7 11:56:45 octopi kernel: [ 1483.542567] be60: b2de3cc3 b2de3c80 b5b2bee4 b5b2be78 7f259ea4 7f259da4 b529a8c0 b2e8e780
Mar  7 11:56:45 octopi kernel: [ 1483.552898] be80: b6fc0580 00000002 00000002 00000013 00000000 00000001 00000000 00000000
Mar  7 11:56:45 octopi kernel: [ 1483.563230] bea0: e5ef7b10 6c77669d 00000030 00000000 00000000 00000000 80569d94 b2c36098
Mar  7 11:56:45 octopi kernel: [ 1483.573562] bec0: b5b15a00 b6fc0094 b6fc0080 b6fc3f00 00000000 00000000 b5b2bf24 b5b2bee8
Mar  7 11:56:45 octopi kernel: [ 1483.583906] bee0: 8003dbf8 7f259e48 b6fc0080 b6fc0094 b5b2a000 00000008 b5b15a00 b6fc0080
Mar  7 11:56:45 octopi kernel: [ 1483.594236] bf00: b5b15a18 b6fc0094 b5b2a000 00000008 b5b15a00 b6fc0080 b5b2bf5c b5b2bf28
Mar  7 11:56:45 octopi kernel: [ 1483.604569] bf20: 8003dec4 8003dab0 8003de70 80818900 00000000 00000000 b5978080 b5b15a00
Mar  7 11:56:45 octopi kernel: [ 1483.614898] bf40: 8003de70 00000000 00000000 00000000 b5b2bfac b5b2bf60 8004364c 8003de7c
Mar  7 11:56:45 octopi kernel: [ 1483.625229] bf60: b5b2bf94 00000000 8004a260 b5b15a00 00000000 00000000 b5b2bf78 b5b2bf78
Mar  7 11:56:45 octopi kernel: [ 1483.635560] bf80: 00000000 00000000 b5b2bf88 b5b2bf88 b5978080 80043560 00000000 00000000
Mar  7 11:56:45 octopi kernel: [ 1483.645893] bfa0: 00000000 b5b2bfb0 8000faf8 8004356c 00000000 00000000 00000000 00000000
Mar  7 11:56:45 octopi kernel: [ 1483.656229] bfc0: 00000000 00000000 00000000 00000000 00000000 00000000 00000000 00000000
Mar  7 11:56:45 octopi kernel: [ 1483.666561] bfe0: 00000000 00000000 00000000 00000000 00000013 00000000 55555575 55555755
Mar  7 11:56:45 octopi kernel: [ 1483.676941] [<7f12eb48>] (cfg80211_get_bss_channel [cfg80211]) from [<7f12fa18>] (cfg80211_inform_bss_width+0x70/0x234 [cfg80211])
[... cut ...]
Mar  7 11:56:46 octopi kernel: [ 1483.811288] [<8004364c>] (kthread) from [<8000faf8>] (ret_from_fork+0x14/0x3c)
Mar  7 11:56:46 octopi kernel: [ 1483.811294] Code: e24cb004 e52de004 e8bd4000 e59032c0 (e5130014) 
Mar  7 11:56:46 octopi kernel: [ 1483.811297] ---[ end trace b08645ac099d9c90 ]---
Mar  7 11:56:46 octopi kernel: [ 1483.811299] Fixing recursive fault but reboot is needed!
```

Big kernel oops there. Apparently in the wifi driver. Did I mention that I have a feeling that the drivers for the included wifi chipset aren't quite there yet?

A-Nesh, 2016-03-07:
@foosel do you think that using a USB WIFI adapter will solv the issue?

foosel, 2016-03-07:
Absolutely no idea. Might be worth a try. Maybe even blacklisting the driver of the built-in wifi.

What will definitely solve a lot though is the RPi people fixing their drivers ;)

A-Nesh, 2016-03-09:
I can confirm now that blacklisting the `brcmfmac` driver solvs the issue for me. It now works flawlessly with a USB WiFi adapter.

foosel, 2016-03-09:
@A-Nesh I forwarded this thread to someone at raspberry pi, the engineering team suggested a different power supply, replied that you already tried that, they also suggested to try stock raspbian - just for the record, when you talked about the "full image" further up, did you mean stock raspbian/noobs?

A-Nesh, 2016-03-09:
@foosel I have used this:
RASPBIAN JESSIE
Full desktop image based on Debian Jessie
Version:February 2016
Release date:2016-02-26

I have tested 4 different power supplies, 2 SD cards, tested setting `max_usb_current=1`and `wireless-power off`... it freeze after a while with and without 3D printer connected. 

Since i have blacklisted `brcmfmac` it printed flawlessly with 1.0A chromecast power supply while running `hardinfo` :)

techied, 2016-03-10:
;D so happy I discovered this thread
Can confirm that latest octopi (not nightly) does not work on my brand-new pi 3 and stock psu
Will follow directions here and either use nightly build or Jessie =D

A-Nesh, 2016-03-10:
i have now switched to octopi nightly build and blacklisted `brcmfmac`. It has now been printing for two hours without problems.

techied, 2016-03-11:
It works with the latest nightly build (released on the ninth of March) 
Haven't tried printing yet, but thanks for the help =D

techied, 2016-03-11:
Finished a print on the nightly build no problem :D 
I used the stock wifi and camera module, and a 2.4amp psu
no crashes or anything :)

A-Nesh, 2016-03-11:
Yeah, it just proves that i have some kind of hardware issue with my WiFi chip.

picclock, 2016-03-15:
Brilliant helpful thread. Same issues with RPi3, been driving me mad. Have a wifi dongle but how do you blacklist the brcmfmac file, or is there an image with it blacklisted?

foosel, 2016-03-15:
@picclock have you the same crashing issues as @A-Nesh? You definitely should verify first that it isn't just your Wi-Fi going to sleep by disabling it's power management as described [above](https://github.com/guysoft/OctoPi/issues/207#issuecomment-192938770). Only if that doesn't work should you test if black listing the module works. To black list it,

foosel, 2016-03-15:
Argh, misclicked, sorry

picclock, 2016-03-15:
Hi Foosel, Thx for the help. I will do the wifi power bit. I can use a LAN connection to edit it as the wifi is all over the place. Am running 2016-02-26-octopi-jessie-lite-0.13.0.img. 
Many Thx

A-Nesh, 2016-03-15:
Yes, i have tried to disable wifi power management with no sucess. It completely freezes, does not recognize monitor and keyboard. The only solution i found working for me is blacklisting `brcmfmac `in `/etc/modprobe.d/raspi-blacklist.conf` 

On the bright side OctoPi nigtly build works great, has been printing with it for ower a week now.

picclock, 2016-03-15:
@ A.Nesh
Exactly the same experience. Followed instructions to disable wifi power mgmt and no wifi. Plugged in LAN cable, ssh, restored backup interfaces file, blacklisted driver and plugged in wifi adapter. Rock solid. !!!!  

Clearly some issues that need sorting.

Many , many thx

best regards

picclock

foosel, 2016-03-15:
@picclock I'm still not 100% sure what problem exactly you are experiencing.

Did you have kernel panics like @A-Nesh (full out system crashes) or not?

There are two issues here:

  1. One manifests as the Pi still being alive but the wifi going down after some idle time due to power management kicking in (and being broken). That can be fixed by disabling power management by adding `wireless-power off` to the wifi config in `/boot/octopi-network.txt`. Apparently there is also a fix available through `rpi-update` or possibly even `apt-get upgrade` by now, but personally I'd prefer a fixed raspbian base image...

  2. The other is a severe bug in the driver causing a full out kernel panic in some cases, effectively locking down the system (and causing such nice things as prints stopping and heaters staying on). That manifests as the Pi not even reacting to a wired connection, keyboard, monitor etc but should hopefully be visible in `/var/log/syslog`. This can be solved by blacklisting the `brcmfmac` kernel module, but that also removes the possibility to use the integrated wifi at all (you need to use a wired connection then, or an external wifi dongle). It's a bit like solving an issue with a home appliance not by repairing it but throwing it on the trash and buying a new one - it's not a fix.

I'm asking to be very specific as to what issue specifically you are seeing (one or two or both), because the RPi people I've been talking to over some hops in between appear not be convinced yet that there are more issues than the power management related one (number 1), so any proofs for the existence of issue number 2 (in the form of hardware setup used + syslog with kernel panic etc) would help to hopefully raise awareness and get this issue addressed as well.

And just for the record, issue number 2 is nothing that can be addressed through OctoPi, it's something that needs to be fixed in the kernel driver for the wifi chipset, so something that needs to be done by the people maintaining that driver.

picclock, 2016-03-15:
Hi Foosel. Thx for your efforts. I disabled the wifi power save as in 
http://www.modmypi.com/blog/disable-wifi-power-management
After this the pi would not connect to the router. It could not be seen at all. I then replaced the file I had edited with the backup, and blacklisted the brcmfmac file and it worked and has been running without issues since. 

It may be that the wifi power mod is incorrect, or, because I deleted the file contents and replaced them with a paste from the web page, that I may have deleted something else which was needed.

The initial problems were that wifi connected intermitantly, then would no longer connect at all. 

If you wish me to run any tests to diagnose I wil be happy to do so, but please make it simple !!

Best Regards

picclock

picclock, 2016-03-15:
@ Foosel
OK, I just disabled the wifi power ( in the octopi-network, added  wireless-power off) and it seems good so far without the dongle. Will keep you updated if it fails.

Best Regards

picclock

foosel, 2016-03-15:
@picclock then blacklisting the module is possibly not necessary

----

#### General instructions on how to test for wifi issues with the RPi3 here

Download and flash a fresh OctoPi *0.13 nightly* image (0.12 doesn't work and will give you the "rainbow image"). Configure your wifi credentials while still using the SD card as a thumb drive while still in the computer you flashed it from by editing `octopi-network.txt` in the root of that thumb drive. Make sure your text editor doesn't mangle the quotes around the ssid and passphrase (especially TextMate on Mac appears to love to do that, you want " or ' if either ssid or passphrase contains spaces, nothing else!). Plug the SD into the Pi, do NOT attach a wifi dongle. Boot the Pi, verify that it connects to your network. Then observe its long time behaviour while using it (e.g. printing stuff).

If an issue arises, is it

a) the Pi stopping to react to network access via wifi but still apparently running (e.g. if a print is running it continues, a keyboard and monitor still work, ...) or

b) the Pi stopping to react to network access and anything else, apparently crashing (e.g. if a print is running it stops, a keyboard and monitor don't still work, ...) or

c) nothing of the above

**If a)**, does this solve the issue?

  1. Shutdown the Pi, unplug the SD, plug it into your computer's card reader
  2. Edit `octopi-network.txt` again
  3. Turn

     ```
     ## WPA/WPA2 secured
     iface wlan0-octopi inet manual
         wpa-ssid "YOUR SSID"
         wpa-psk "YOUR PASSWORD"
     ```

     into 

     ```
     ## WPA/WPA2 secured
     iface wlan0-octopi inet manual
         wpa-ssid "YOUR SSID"
         wpa-psk "YOUR PASSWORD"
         wireless-power off
     ```
     and save.
  4. Put the SD back into the Pi, reboot, observe

**If b)**, please provide your `/var/log/syslog` file.

**If c)**, sounds like a so far unknown issue.

If you are only experiencing a), the solution to a) should work. If you are not seeing your Pi3 connect  to your network after applying the solution to a), something else is broken in your configuration that has nothing to do with either a) or b) (probably a mistyped password, mangled quotes or something similar).

Please do not try to apply other steps outlined in other blog posts around the internet, follow the steps outlined in this thread. Give precise feedback as to what exact behaviour you are observing at all times.

ghollingworth, 2016-03-15:
Hi,

Gordon Hollingworth here from Raspberry Pi.  The WiFi power management issue has now been fixed, apt-get update / upgrade on a Raspbian distribution will fix this for you.  I'd suggest pulling that kernel into the OctoPi distribution.

The driver crashing issue is specific to this case only.  We're not seeing this anywhere else so it is somewhat strange (not saying it isn't a problem just that it's going to be hard to understand / diagnose!)

Could do with answers to the following:

1) If you return to Raspbian Jessie and sudo apt-get upgrade / update
2) edit /boot/config.txt and change arm_freq to 600 (i.e. force underclock just in case)
3) Does this still cause the problem
4) Other possibility is that it's something to do with the WiFi access point settings...  Can you try forcing the channel to somewhere in the middle of the range and turn off encryption for a little while to see if's to do with that?

Gordon

A-Nesh, 2016-03-15:
@ghollingworth i'll try it later today and get back to you

foosel, 2016-03-15:
@ghollingworth thanks for reaching out. We can add the kernel update step to the OctoPi build script, but would prefer to base OctoPi on a Raspbian release that already includes that. Do you know if a new stable image release is planned soon? If not we'll have to go with adding the upgrade (in the hopes that this works with our qemu based build).

ghollingworth, 2016-03-15:
We are planning on a new update soon, but currently no date, should be in
the next week or so�

Gordon

foosel, 2016-03-15:
Thank you, that's something to go on.

seabas917, 2016-03-15:
Hey guys, 
Am I crazy or is the latest published nightly (2016-03-15_2016-02-26-octopi-jessie-lite-0.13.0) missing its Octoprint install?

On the other hand it looks like after ssh-ing in wifi credentials, the onboard wifi is working really well! Same ping quality as my laptop, in this build. I'll leave ping test up all night to see if this is stable. I'll do a manual Octoprint install after seeing if the system stays up.

Thanks Foosel and Gordon! 
Amazing work from both of you!

Side note: No surprise here, but the AdafruitLCD install script in Nightly(03-15) for the OctoPanel, will crash the Pi3 system on next reboot and not recover.. 

The boot/octopi-network file isn't working, so i suggest manual setup for onboard wifi:
`sudo nano /etc/wpa_supplicant/wpa_supplicant.conf `

and adding a WPA2 network:
```
network={
    ssid=“*****”
    psk="*****"
}
```
Reboot Wifi:
```
sudo ifdown wlan0
sudo ifup wlan0
```
Then confirming Wifi IP with : `ifconfig`

guysoft, 2016-03-16:
Yes! yesterday nightly is bad, don't use it

[45 more replies]

closed 2016-05-01 by guysoft