guysoft/OctoPi #208: Stuttering. Maybe direct serial connection?

nickbardom, 2016-03-02:
Hi, I decided to try OctoPi with my Raspberry PI B version and the latest Marlin release.

I am not using a webcam and I am using a wired connection, but I can see the printer stutters when the Gcode have complex movements. It only stutters using OctoPI. Using direct computer connection or direct SD print works fine. I can see it is less than 20% of processing power all the time. So I think the problem is the USB cannot keep sending the Gcode as fast as it should. I tried to reduce the baud rate from 250000 to 115200, but the problem is the same.

So what do you think about connecting the Raspberry serial UART directly to the UART of the arduino using a logic level shifter?

Is there way I can do it by now in the config files?

nophead, 2016-03-02:
I do that on one of my machines but it doesn't get around the basic problem the OctoPrint serial comms isn't fast enough for models with short segments.

It doesn't require any changes to OctoPrint other than configuring it to use /dev/ttyAMA0 as the serial port in the settings. I use a 2K2 and a 3K3 potential divider on the receive line to lower it to 3V and a 1K in series with the transmit line to prevent current flow when the board is powered down and RPi isn't

I changed the Marlin configuration to use the second serial port but I still need to update the firmware via USB as I haven't bothered changing the bootstrap.

On the RPi I had to configure it not to use the serial port as a boot console and also not spawn a terminal process on it. I found that by Googling.

By default the maximum  speed is 115200 and that seems to slightly faster than USB at 250000 but not much. With later Raspbian (wheezy I think) it is possible to change the RPI serial port clock speed and get higher baud rates. I have had an RPI transmitting at 1Mb but I haven't tried it with a printer. At some point the firmware will be flooded with interrupts and fall over.

The serial baud rate isn't currently the limiting factor though, see #450.

nickbardom, 2016-03-02:
Hi nophead. Thanks for replying I will follow the thread you suggested. Thanks.

nickbardom, 2016-03-05:
Hi again,

I just made some new tests with Repetier-server. Same results. After that I tried the Repetier firmware. And I got full speed printing. No stuttering. So as the Repetier protocol takes less instructions, it works. Does OctoPi supports the Repetier protocol?

foosel, 2016-03-05:
OctoPi is an sd card image, it doesn't support any protocols. Octo*Print* (the server/ui thing you are using for controlling your printer) does, but doesn't support the binary repetier protocol. Should be possible to write a plugin for it, but nobody has done that so far as far as I know.

nickbardom, 2016-03-05:
Hi foosel,

Sorry for messing with the names. And thanks for replying!!!

foosel, 2017-02-28:
Closing, not an OctoPi issue. If anyone reading this is interested in writing a plugin to get the binary Repetier protocol working from within OctoPrint, best get in touch via a [Brainstorming ticket on OctoPrint's bug tracker](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#issues-tickets-however-you-may-call-them), or via the [mailing list](https://groups.google.com/group/octoprint).

closed 2017-02-28 by foosel