guysoft/OctoPi #209: OctoPi File Upload Problem

135erHeli, 2016-03-02:
I got Problem with a clrealy new Octoprint Image.
Octoprint Master Brunch 1.2.9
Raspberry PI B+
Repetier Firmware 0.92
Firefox and Safari

If i will Upload i got the Message Upload fails, but in Refresh the file ist Uploaded and Printable.But if ill Print them at 99% the Print Stops due a missing Line Number.

[image]
[serial.pdf](https://github.com/foosel/OctoPrint/files/154547/serial.pdf)
[octoprint.pdf](https://github.com/foosel/OctoPrint/files/154564/octoprint.pdf)
#

nophead, 2016-03-02:
It looks like this line is your problem: -
```
2016-03-02 08:40:50,304 - SERIAL - DEBUG - Send: to release some of the pressure
```
I think it is meant to be a comment but gets sent with no line number.
```
N10145 G1 Z10 F{travel_speed}*18
```
I think {travel_speed} should have been replaced with a number by your slicer. 

So it looks like your end gcode section is faulty.

135erHeli, 2016-03-02:
Ill Sliced with Slicer, if ill Print this File Directly from SD it runs without Problems till End

foosel, 2016-03-02:
Sorry, but this is the wrong bug tracker, it's the one for OctoPi (the sd card image), not OctoPrint (the server software).

[Please report this in the correct bug tracker, thank you](https://github.com/foosel/OctoPrint/blob/master/CONTRIBUTING.md#issues-tickets-however-you-may-call-them).

closed 2016-03-02 by foosel

135erHeli, 2016-03-02:
But ill Use OCTOPI Image on a RaspberryPI so im wrong here?

foosel, 2016-03-02:
Yes. For the same reason why you would be wrong in the Raspbian bug tracker if you had an issue with the bundled node-red software. Our why you wouldn't call your car's manufacturer to fix a problem with your mobile phone that occurred while you happened to sit in the car.

OctoPi is an image which already has OctoPrint, a webcam server and some other useful stuff installed so you don't have to set this up yourself. But you are having issues with one of the included components (OctoPrint), not the image (OctoPi). So you should report this issue in OctoPrint's bug tracker, not OctoPi's.

135erHeli, 2016-03-02:
Ahh OK Complicated but ill Posted also there but the Bot says something should be missing. Maybe a German Part of it cause my English is the best :)