guysoft/OctoPi #210: Webcam Problem Logitech C920

135erHeli, 2016-03-03:
I use a Logitech C920 Webcam.If ill start up Octopi the Webcam is running and ill see the Picture.After a while the Picture ist gone but the Process is still alive.
Using latest Stable OctoPi image with latest Octoprint Master Update 1.2.9

Here the Output from the webcamd.log
`pi: Starting USB webcam
Running ./mjpg_streamer -o output_http.so -w ./www -i input_uvc.so -r 640x480 -f 10
MJPG Streamer Version: svn rev: Unversioned directory
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 10
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
Unable to set format: 1196444237 res: 640x480
 Init v4L2 failed !! exit fatal 
 i: init_VideoIn failed
pi: Starting USB webcam
Running ./mjpg_streamer -o output_http.so -w ./www -i input_uvc.so -r 640x480 -f 10
MJPG Streamer Version: svn rev: Unversioned directory
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 10
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
Unable to set format: 1196444237 res: 640x480
 Init v4L2 failed !! exit fatal 
 i: init_VideoIn failed
pi: Starting USB webcam
Running ./mjpg_streamer -o output_http.so -w ./www -i input_uvc.so -r 640x480 -f 10
MJPG Streamer Version: svn rev: Unversioned directory
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 10
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
Unable to set format: 1196444237 res: 640x480
 Init v4L2 failed !! exit fatal 
`
When i replug to USB it works

`pi: Starting USB webcam
Running ./mjpg_streamer -o output_http.so -w ./www -i input_uvc.so -r 640x480 -f 10
MJPG Streamer Version: svn rev: Unversioned directory
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 10
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path...: ./www/
 o: HTTP TCP port.....: 8080
 o: username:password.: disabled
 o: commands..........: enabled
pi@octopi /var/log $ 
`

135erHeli, 2016-03-05:
Fixed

closed 2016-03-05 by 135erHeli

foosel, 2016-03-05:
Since there might be other people finding this in the future via Google and such, it would be great if you could also say how you fixed it.

135erHeli, 2016-03-05:
Did a apt-get dist-upgrade instad of apt-get upgrade to use all new packages