guysoft/OctoPi #211: Wi-Fi doesn't reconnect automatically

chaseacton, 2016-03-03:
To reproduce:
1) Have OctoPi running normally. Make sure you can connect via Wi-Fi
2) Reboot router, access point, etc.
3) Notice OctoPi will not automatically reconnect to the network. A power cycle is required to be able to connect again

Salandora, 2016-03-04:
Which version of OctoPi do you use?
What WLAN adapter do you use?

I had the same problem once.
I was able to fix it.

chaseacton, 2016-03-04:
@Salandora Latest version with the WiPi adapter.

markwal, 2016-03-04:
@chaseacton Out of curiosity, which of the latest versions do you mean?  Latest jessie, 0.12 (latest stable), latest banana pi, etc.?

chaseacton, 2016-03-04:
@markwal  I'm using OctoPi 1.2.9 from https://github.com/foosel/OctoPrint/releases/tag/1.2.9 on the Raspberry Pi 2.

markwal, 2016-03-04:
@chaseacton No that's the version of OctoPrint you are using and OctoPrint doesn't know anything about your wifi.  What is more interesting here is the version of the OctoPi image that you downloaded to put on your SD card.  The OctoPi image on the other hand has a particular version of raspbian, included drivers, etc.

chaseacton, 2016-03-04:
@markwal It's 0.12, the latest Raspbian image from https://octopi.octoprint.org/latest

Salandora, 2016-03-05:
Would be interesting to have some sort of logs, e.g. what is "dmesg | tail" saying after your router turns off...
I'll download the image flash it and check if this happens for me toowith my 2 Wifi adapters...
Some information: what router/access point do you have?

Salandora, 2016-03-05:
Ehh okay, I got why your Wifi isn'T reconnecting, the OctoPi v0.12 doesn't include my wifi patch -.- ...

If you want to get it working here is the instruction:

ssh into your pi (or use a monitor and keyboard :-))
then type the following (or copy paste it):
`mv /etc/ifplugd/action.d/ifupdown /etc/ifplugd/action.d/ifupdown.original`
`ln -s /etc/wpa_supplicant/ifupdown.sh /etc/ifplugd/action.d/ifupdown`

after that, please restart your pi, and test it, and report back :-)

foosel, 2016-03-05:
> Ehh okay, I got why your Wifi isn'T reconnecting, the OctoPi v0.12 doesn't include my wifi patch -.- ...

Of course it doesn't, your patch PR came after release of 0.12 ;)

chaseacton, 2016-03-06:
@Salandora Awesome, that worked.

@foosel Gina, can we expect this fix in the next release? :)

Salandora, 2016-03-06:
@foosel, yeah I noticed that too...

pjvpjv, 2016-03-12:
I have a hunch that this patch not being in 0.12 is one reason why there are a number of people having issues with initial wifi configuration in 0.12 using octopi-network.txt.  I use a Meraki access point and for some unknown reason the first connection attempt upon raspberry pi boot fails... every time.  As soon as I apply this patch, it connects perfectly on boot.  Without @Salandora 's patch, it never connects without manually running `ifup --force wlan0` after boot.  With @Salandora 's patch, it connects on initial boot perfectly.

So, whenever you hear about someone struggling to get initial wifi settings to work (there are a number of threads talking about this with 0.12), you might consider this as a possible fix for them.  I'd love to see this in the next release (perhaps even a stopgap 0.12.1 release?), or else a certain subset of people trying OctoPi for the first time will continue to throw up their hands in frustration because wifi settings won't work for them.

guysoft, 2016-03-12:
We have sevral major issues addressed with the upcoming 0.13.0. Its also the longest we had between releases, so I feel your pain. We are mostly hanging now because Rpi foundation are working on a power fix for the current Rpi 3. Once that is out I think we will release 0.13.0 (with @foosel 's go ahead of course). In the meantime hang in here and use nightly.

techied, 2016-03-23:
@guysoft Are those power fixes software-based? If not, I have to get a new Pi3 :(

markwal, 2016-03-23:
The wifi power problem is a software problem.  It's fixed in the new jessie image.

The kernel panic too is likely to be able to be fixed in software.  Though it maybe that it is caused sagging voltage due to increased draw from the wifi module, in which case, if you're experiencing it you'll likely only need a new power adapter, not a Pi3.

guysoft, 2016-05-01:
Fixed in 0.13.0

closed 2016-05-01 by guysoft