guysoft/OctoPi #212: image fails to resize during first boot

mtowara, 2016-03-05:
After flashing the newest jessie nightly i ran into the following issue:
The first boot got stuck (waited for >30 min), after rebooting it went straight to login, however I then noticed that 0 Byte were left on the sd card. After manually deleting some files I was able to resize the partition using raspi-config.
According to #166 and octopi.txt i think it should have been resized automatically, so maybe the initial image size has become slightly too small?

Hardware: Raspberry Pi 3, I set a WiFi connection in octopi-network.txt before first booting, maybe that somehow increases used space?

guysoft, 2016-03-06:
No Rpi3 here, can't test. Waiting for it, and since Israel post office collapsed it might even take a month.

foosel, 2016-03-06:
Cannot verify, just tested the nightly build from 2016-03-02 on a pi3, flashed to an 8GB Class 10 Micro SD card from Transcend. Image resized without issue on first boot.

Maybe something went wrong during the initial flash? What nightly did you use, maybe there was something wrong with it? Just for the record, the initial resize script uses `raspi-config`, so it does exactly the same thing as performing the resize through the `raspi-config` menu.

mtowara, 2016-03-09:
Turns out I cant reproduce this specific issue either, but I keep getting strange issues. I think something is wrong with either my SD card, that adapter or the way XUbuntu handles unmounting the image.

Once I got an working image this time the first boot got stuck after "stopped hostname service", but after powercycling once this time resize and everything else went through.

Nightly was 2016-03-02_2016-02-26-octopi-jessie-lite-0.13.0

guysoft, 2016-03-10:
Check you are using at least 2A of power and a good power cable. Sounds
like a power issue.

3driffs, 2016-03-12:
Same issue here on my 3 - full screen rainbow square (and small one in top right).  Tried two different 2.5a supplies.  But I'm trying with the last stable release - will update with latest nightly and see if it works.

mtowara, 2016-03-14:
I'm using the official RPi 2A Supply and am not getting the small rainbow square, don't think its power related in my case as everything is rock solid once i manage to get it booting once.

For @3driffs it certainly sound like an power issue, is your printer already connected and maybe draining a lot of power from the USB ports?

foosel, 2016-03-14:
No, for @3driffs it sounds like the expected behaviour when trying to boot on a Pi3 with the current stable 0.12 image, since that doesn't yet know what to do with the Pi3. Pi3 needs one of the nightly builds. And we can't push a new stable release with Pi3 support yet since the WiFi drivers for the internal WiFi interface are currently buggy on the official Raspbian image and causing various issues. See #207

guysoft, 2016-05-01:
Fixed in 0.13.0

closed 2016-05-01 by guysoft