guysoft/OctoPi #213: "Load and print" button broke in latest update

giantpune, 2016-03-18:
I have octopi running with the gpx plugin.  Just now I received an update for "OctoPrint: 1.2.10".  After that update, the octopi restarted.  Now I can list the gpx files on my sd card as before, but clicking the little "Load and print" button does nothing.  I have to press the buttons on the front of my printer and browse to the file to print it like some sort of caveman.

guysoft, 2016-03-18:
Hey,  
You better post this in the OctoPrint issue tracker. 
https://github.com/foosel/OctoPrint/issues

giantpune, 2016-03-18:
I'll try, but their issue tracker says not to post issues if they have to do with octopi.

giantpune, 2016-03-18:
@guysoft is there an easy way in octoprint to roll back to 1.2.9 until the fix is identified and released by whoever needs to fix it?

guysoft, 2016-03-19:
1. This is an OctoPrint issue. The update system belongs to OctoPrint. Therefore its not an OctoPi issue.
2. I am not sure what the effects to rolling back would be, I suggest you ask that there too.

jminardi, 2016-03-21:
@giantpune To roll back octoprint you just need to checkout whichever version you want in the OctoPrint src directory.

ssh into the pi
`cd /home/pi/OctoPrint && git checkout 1.2.9`

guysoft, 2016-05-01:
Looks like its fixed on OctoPrint's end

closed 2016-05-01 by guysoft