guysoft/OctoPi #214: locale.Error: unsupported locale setting

jminardi, 2016-03-19:
Recently my OctoPi builds have been failing with a locale error. This is with no change on my end, one day it worked, the next it was failing.

Here is the error:

```python
+ sudo -u pi /home/pi/oprint/bin/pip install ndg-httpsclient pyasn1
sudo: unable to resolve host ubuntu
Traceback (most recent call last):
  File "/home/pi/oprint/bin/pip", line 11, in <module>
    sys.exit(main())
  File "/home/pi/oprint/local/lib/python2.7/site-packages/pip/__init__.py", line 215, in main
    locale.setlocale(locale.LC_ALL, '')
  File "/home/pi/oprint/lib/python2.7/locale.py", line 579, in setlocale
    return _setlocale(category, locale)
locale.Error: unsupported locale setting
```

guysoft, 2016-03-19:
What image are you building against?
Because I don't get this error on raspbian jessie but I do get it in the bananapi variant. It seems to happen because the locale used is missing in the image

jminardi, 2016-03-19:
I have been using 2016-02-09-raspbian-jessie

jminardi, 2016-03-19:
I had previously been getting a bunch of perl locale warning but they didn't seem to cause any problems:

```
perl: warning: Setting locale failed.
perl: warning: Please check that your locale settings:
	LANGUAGE = "en_US.UTF-8",
	LC_ALL = "en_US.UTF-8",
	LANG = "en_US.UTF-8"
    are supported and installed on your system.
perl: warning: Falling back to the standard locale ("C").
```

jminardi, 2016-03-19:
Just tried2016-03-18-raspbian-jessie 

same problem

guysoft, 2016-03-19:
Can you try using the vagrant build method provided in the readme? That should make sure your build environment is the same as mine.

jminardi, 2016-03-19:
After more digging I figured the error must be in a recent pip. I tried installing pip 8.0.3 instead of the latest and it fixed the problem.

I can try the vagrant build and will report back

guysoft, 2016-03-20:
Confirmed, it started happening on my builds too.

Can you post your current fix.

guysoft, 2016-03-20:
Adding ``export LC_ALL=C`` at the start of the chroot script seems to fix it, running a new nightly build to test it and if that works it would be committed in an hour or so.

commit 0f1a0759 (2016-03-20): Fix #214 pip install locale bug
closed 2016-03-20 by guysoft