guysoft/OctoPi #215: Watterott RPI: System does not restart after enable-rpi-display

RollingGecko, 2016-03-20:
Pi2 B with Watterott 2.8'' Display
I installed the latest Octopi and expanded the file system. After I executed the script enable-rpi-display and made a shoutdown -r now the system only shows the multicolor test screen. No startup.

guysoft, 2016-03-20:
Suggest you post to https://github.com/jonaslorander/OctoPiPanel/issues @jonaslorander seems to handle that stuff much better.

ChrisWeiss, 2016-05-05:
FWIW - I see the same thing with the Pi2 and enable-adafruit-pitft. The Kernel created by the script does not appear compatible with the Pi2. Digging further as the two scripts appear to do radically different things.

lrvick, 2016-08-29:
Same issue with me. Ran enable-adafruit-pitft on my pi 3 and it never comes back up. Had to re-flash :(

guysoft, 2019-07-02:
OctoPiPanel is legacy, closing.

closed 2019-07-02 by guysoft