guysoft/OctoPi #216: not recognizing RPI camera

dsolthesoul, 2016-03-21:
hey..
got the RPI camera moudle and plug it in... but for some reason in the octopi it wont show!

tryed to SSH and used "raspistill" and everything works fine and i get a picture

this is my octopi.txt file

camera="raspi"
camera_raspi_options="-x 1920 -y 1080 -fps 30"

any idea why? or how to fix that ?

guysoft, 2016-03-21:
What OctoPi version are you using?
What RaspberryPi are you using?

dsolthesoul, 2016-03-21:
OctoPrint: 1.2.10 
RaspberryPi 1 (2011)

foosel, 2016-03-21:
Octo**Pi** version. You gave the Octo**Print** version.

`cat /etc/octopi_version`

dsolthesoul, 2016-03-21:
ohh.. didnt know its different

Pi version - 0.12.0

WZ9V, 2016-06-30:
Are you sure you have the cable plugged in the right way?  It's easy to get that ribbon cable backwards and that would cause what you are seeing.  I think you can also go straight into the webcam server as octopi.local:8080 which is helpful for troubleshooting camera issues.  Once you have that working it's usually all good in OctoPrint.

guysoft, 2016-06-30:
Please use 0.13.0

dsolthesoul, 2016-06-30:
yea.. it an old problame, been fixed for me when 0.13.0 come out

foosel, 2017-02-28:
Problem reported as fixed.

closed 2017-02-28 by foosel