guysoft/OctoPi #217: Raspi3 camera not detected

samjhill, 2016-03-28:
OctoPi version: 0.13.0
uname -a: Linux octopi 4.1.20-v7+
Raspberry Pi Model 3 revB

Console output:
`raspistill -v -o test.jpg

raspistill Camera App v1.3.8

Width 2592, Height 1944, quality 85, filename test.jpg
Time delay 5000, Raw no
Thumbnail enabled Yes, width 64, height 48, quality 35
Link to latest frame enabled  no
Full resolution preview No
Capture method : Single capture

Preview Yes, Full screen Yes
Preview window 0,0,1024,768
Opacity 255
Sharpness 0, Contrast 0, Brightness 50
Saturation 0, ISO 0, Video Stabilisation No, Exposure compensation 0
Exposure Mode 'auto', AWB Mode 'auto', Image Effect 'none'
Metering Mode 'average', Colour Effect Enabled No with U = 128, V = 128
Rotation 0, hflip No, vflip No
ROI x 0.000000, y 0.000000, w 1.000000 h 1.000000
mmal: mmal_vc_component_enable: failed to enable component: ENOSPC
mmal: camera component couldn't be enabled
mmal: main: Failed to create camera component
mmal: Failed to run camera app. Please check for firmware updates`

So far, I've tried:

1) Adding this to my /boot/config.txt
`cma_lwm=
cma_hwm=
cma_offline_start=`

2) Changing my blacklist.conf
`#blacklist spi-bcm2708
blacklist i2c-bcm2708`

3) Changing my /etc/modules
`w1-therm
w1-gpio pullup=1
i2c-dev
i2c-bcm2708
spi-bcm2708
snd-bcm2835
bcm2708_wdog`

foosel, 2016-03-29:
Raspicam works just fine in OctoPi 0.13 on a Pi3 (just verified as I'm writing this) without having to configure anything.

`raspistill` will not be able to work when `mjpg-streamer` is already running and accessing the camera (and will report the *exact* error like the one you see up there in such a case), and OctoPi has `mjpg-streamer` autostart on boot. So that a call to `raspistill -v -o test.jpg` fails with a `ENOSPC` error is actually expected behaviour when the camera *was* detected properly.

Elaborate on the exact issues you are seeing. You never stated what you mean with "not detected", you only gave a `raspistill` command that is expected to fail if the camera in fact *is* detected and claimed by `mjpg-streamer`, so that's a bit nondescriptive when it comes to what your actual problem is. If your camera's led comes on and you have an image on the Control tab within OctoPrint, everything is just working as it should. If the camera light comes on but you don't see anything in the control tab, it's most commonly caused by a [misconfiguration by the user](https://github.com/foosel/OctoPrint/wiki/FAQ#how-do-i-have-to-configure-the-webcam-urls-in-octoprint-so-that-everything-will-work). If the camera light doesn't come on, verify you've wired up the camera correctly and ideally test it with another setup to make sure it isn't broken. Never connect or disconnect the Raspicam while the Pi is powered up!

samjhill, 2016-03-29:
Thanks for the reply! The red LED is on, but I don't see anything in `:8080?action=stream` or the control tab. I just ordered another camera module just on case mine is broken - I'll let you know if that fixes it. I also looked through the configuration link, but didn't immediately see anything that was different from my config. Thanks again!

foosel, 2016-03-29:
> The red LED is on, but I don't see anything in :8080?action=stream or the control tab.

Then the camera is most probably fine, or it wouldn't be running. Verify the URLs, make sure you access OctoPrint through port 80 (otherwise `/webcam/?action.stream` won't work on the control tab), etc. 

`http://octopi.local/webcam/?action=stream` should work in Chrome, but not all browsers support mjpg directly and only in an img tag (don't ask me why), so maybe rather test `http://octopi.local/webcam/stream.html`.

In any case, my money would be on some network or configuration issue here, not your Pi camera.

samjhill, 2016-03-31:
Still waiting on the second camera to come in, but I also wiped the card and installed the [most recent nightly build](https://storage.googleapis.com/octoprint/2016-03-31_2016-03-18-octopi-jessie-lite-0.13.0.zip), and re-enabled the camera in raspi-config. No change in symptoms, unfortunately.

samjhill, 2016-04-06:
It was the camera.

closed 2016-04-06 by samjhill