guysoft/OctoPi #218: Yet another Serial Port issue (RPI2 + Da Vinci 1.0 Pro)

BradenMars, 2016-04-01:
Wow. So I have been attempting to setup octopi for quite a while. Basically, when I leave the serial port on "AUTO," I receive the error: `The was an unhandled error while talking to the printer. Due to that OctoPrint disconnected. Error: Failed to autodetect serial port, please set it manually.` 

At first, I thought it was the fact that my wifi card wasn't cutting it as it required too much power. So now, a while later, I have the official WiPi wireless card and am running into the same exact issue. 

While looking for an answer, I came across a page explaining how to check if your RPI is detecting your USB by using the command: `dmesg`   -  I did this and here is what I got (when It reconnected)

`[  684.657916] usb 1-1.5: new full-speed USB device number 10 using dwc_otg
[  684.760622] usb 1-1.5: New USB device found, idVendor=11f1, idProduct=2520
[  684.760649] usb 1-1.5: New USB device strings: Mfr=0, Product=0, SerialNumber=3
[  684.760667] usb 1-1.5: SerialNumber: 3F1AWPUS5TH5AQ0074
[  684.762023] cdc_acm 1-1.5:1.0: ttyACM0: USB ACM device`

So I figured I should use the /dev/ttyACM0 serial port. I tried all baudrates and still have come across nothing. I have tried other serial ports (I cant remember most) but I dont think they really matter. 

It has taken me forever to finally give up and post, so thanks for any replies. Sorry if I forgot any crucial detail.

guysoft, 2016-04-01:
What firmware version do you have installed.  From what I see you need to flash a custom one:
http://www.soliforum.com/topic/9921/offline-printing

BradenJM, 2016-04-01:
I am currently on the latest version of octopi and on both my Printer's firmware and software, but have attempted this on the previous 2 versions of octopi with the same results. I read the forum thread you sent me, but unless I am just completely missing something, I don't see anything about a custom version. Thanks.

guysoft, 2016-04-01:
OctoPrint will not work with the stock firmware. You need to install a diffrent one. See the fourm post I linked to in my previous message, since I have no Da Vinci printer to test. People are reporting success woth repiter firmware.

BradenJM, 2016-04-01:
Hm. I wonder I never came across that. I never thought about reflashing my  Da Vinci due to the fact that most people did it so they could use other filaments and use other software, which, the 1.0 PRO already allows. Perhaps I am wrong and their is more, I will look into it. Thank you!

guysoft, 2016-05-01:
@BradenJM Did you fix this? Since you seem to be thanking people in https://github.com/luc-github/Repetier-Firmware-0.92/issues/151

BradenJM, 2016-05-01:
@guysoft Sadly, no. @luc-github Explained to be that you to re flash the Da Vinci stock firmware with something like reptier. This is because the stock firmware, unlike many others, doesn't actually have a host at all. That is why it's unnecessary to keep it plugged in during printing. Instead, it receives the package (the item to print) and stores in on the SD card. 

So all of this wouldn't matter if you can simply reflash the SD card with reptier, but according to @luc-GitHub referenced me to this post, https://github.com/luc-github/Repetier-Firmware-0.92/issues/142 , where he explains it is incompatible with the PRO. At least last time I checked.