guysoft/OctoPi #219: Support setting fixed IP addresses in 0.13

040medien, 2016-04-11:
Debian Jessie no longer supports static IPs in /etc/network/interfaces, so you can't configure them in octopi-network.txt either. They need to be defined in /etc/dhcpcd.conf

Example:
interface wlan0
static ip_address=192.168.10.99/24
static routers=192.168.10.1
static domain_name_servers=75.75.75.75 75.75.76.76

guysoft, 2017-07-23:
Ok, I got the chance to test and that works, we could export ``/etc/dhcpcd.conf`` to /boot, but I am not sure we should do that.
In the log term its better for people to set their static IPs in router settings and on the Pi, where there is a chance of clashing IPs.

040medien, 2017-07-23:
The default certainly should be DHCP, but manual addresses as an advanced use case make sense. For example I'd like to keep my dynamic IP addresses and my static ones in separate ranges. My router doesn't allow me to configure static IP addresses outside of the range it hands out for dynamic addresses, so the only way to do this is to set it on the client side.

braytonstafford, 2018-01-07:
The static IP for your wlan interface can be configured in */boot/octopi-network.txt* along with the SSID and password for your wifi connection.

Here is an example of a static IP address on a WPA wifi network:

    ## WPA/WPA2 secured
    iface wlan0-octopi inet static
        wpa-ssid "yourWirelessSsid"
        wpa-psk "yourSuperSecureWifiPassword"
        address   192.168.1.2
        netmask   255.255.255.0
        broadcast 192.168.1.1

### Note: the word static at the end of "iface wlan0-octopi inet static" by default the examples in octopi-network.txt say manual.

guysoft, 2019-04-23:
Stetting up static ip for wifi instructions are here:
https://github.com/guysoft/FullPageOS/issues/222#issuecomment-380487942

closed 2019-04-23 by guysoft