guysoft/OctoPi #221: Some settings pages don't render correctly on Macs

040medien, 2016-04-11:
Most of the right hand settings pages remains white until I select some text. This happens on Chrome (no Extensions) an Safari running on OS X 10.11.3.﻿

[image]
[image]

markwal, 2016-04-11:
This, if it is a bug, is in OctoPrint, not in the script that makes a Raspbian image out of it (octopi).  The repo for that is https://github.com/foosel/OctoPrint.

markwal, 2016-04-11:
I do not see this occur on either Chrome or Safari on OS X 10.11.3 in either OctoPrint 1.2.10 or OctoPrint 1.3.0-devel.  Though I do have plugins, in particular, ScreenSquish, which might change this behavior.

guysoft, 2016-05-01:
Not related to OctoPi, use https://github.com/foosel/OctoPrint

closed 2016-05-01 by guysoft