guysoft/OctoPi #222: Editing octopi-network.txt or octopi.img using default text editor on mac mangles encoding

LinkReincarnate, 2016-04-17:
Not an issue with octopi per se but maybe worth making a note of in the documentation.  If you edit either of the config files using the default text editor on mac osx the encoding of the file is broken.  This leads to the scripts being ignored and nothing loading.  I ran into this issue and wanted to help someone avoid it in the future.  TextWrangler (a free app) can edit the files without messing up the text encoding.

guysoft, 2016-04-17:
Its an issue we know of. I just realised its not documented in the wiki at: https://github.com/foosel/OctoPrint/wiki/FAQ  
You are welcome to add that there.

 The main issue is people don't read document, we are not sure how to fix that yet :-/

foosel, 2016-04-17:
Maybe in that specific case it might make sense to add a note to the file itself (in the comments there).

KevMags, 2016-04-24:
OSX TextEdit uses "smart quotes" and rich text by default. In TextEdit preferences select "Plain text" and uncheck "smart quotes".

guillaumemolter, 2016-11-30:
Hi,

I've added a note in : https://github.com/foosel/OctoPrint/wiki/FAQ/#why-does-octopi-not-load-the-wlan.

Following @foosel suggestion I've opened a PR : https://github.com/guysoft/OctoPi/pull/279

Hope this helps!

guillaumemolter, 2016-12-01:
I think this issue is now closed. Octopi cannot change the native behavior of TextEdit. We could eventually open a radar with Apple arguing that opening a .txt file should be "plain text" and without rich formatting options by default but I doubt it would make any difference.

foosel, 2016-12-01:
I agree.

closed 2016-12-01 by foosel

LeonFedotov, 2017-02-16:
I think i know what the problem is, the fact that those files are txt format so osx opens them in the writer by default, this writer adds those quote-marks, the solution is to omit the filetype or change it to .conf or something that wont open the letter writing software @foosel

foosel, 2017-02-16:
And that will then confuse all windows users and go against the existing conventions of configuration files on Raspbian's boot/FAT partition, plus a change in the name of that file will in general yield endless support requests as a consequence due to all those guides around the web that are not in our control referring to a .txt file.

The actual solution would be that the default OSX text editor stopped trying to be fancy and just inserted a `"` when someone presses the `"` key, but of course that would not be "thinking differently" enough I guess.

guysoft, 2017-02-16:
/me just notes he told @LeonFedotov to write that, after he reports the finding to him

LeonFedotov, 2017-02-16:
whatever guys, i'm not here for a flame war - add this as a note to the official tutorial or something, the advanced users wont run into this problem because they have a normal text editor handy, its the newbies that are getting hurt.