guysoft/OctoPi #223: nightly build

prophetmaster, 2016-04-17:
It's normal the latest build have no octoprint installed ?

guysoft, 2016-04-17:
1. No
2. Please try the release candidate.
Download it at:
https://storage.googleapis.com/octoprint/octopi-jessie-lite-0.13.0_RC1.zip
md5:
https://storage.googleapis.com/octoprint/octopi-jessie-lite-0.13.0_RC1.zip.md5

Bananapi M1 (New variants expected at 0.14.0):
https://storage.googleapis.com/octoprint/octopi-bpi-r1-0.13.0_RC1.zip
MD5:
https://storage.googleapis.com/octoprint/octopi-bpi-r1-0.13.0_RC1.zip.md5

prophetmaster, 2016-04-17:
i have downloaded this and tested 3 time not working...

namelivia, 2016-04-17:
I have been experencing issues when the latest nightly builds on a Raspi3. After expanding the filesystem and rebooting the pi there is no webserver or anything on the pi's ip but I can still get into the pi through sshd. Can I do any checks or tests in order to diagnose what is going on?

guysoft, 2016-04-17:
@namelivia do older versions work? sounds like a power supply issue/power supply issue thay caused corruption. The Rpi3 takes more power.

namelivia, 2016-04-17:
@guysoft thanks for pointing me, I've made these tests and got these results:

1.- Switching to a more powerful power source.
2.- Install the RC nightly (2016-04-09_2016-03-18-octopi-jessie-lite-0.13.0.zip) on the sd card
3.- Theres is octopi-network file on the sd, but I add the network on /etc/wpa... etc.. and the rasberry successfully connects to the wifi
4.- I expand the filesystem and reboot, no webserver.

Then I downloaded an older version (2016-04-07_2016-03-18-octopi-jessie-lite-0.13.0.zip). For repeating the same process (with the new powersource).

1.- This time there is an octopi-network file, I set the ssid and psk values there.
2.- Expand, reboot, etc... and now this time the webserver is working :dart: 

So finally I've managed to run octopi on the raspi 3 but no the last nightly. Thanks for pointing me in the right direction.

wiredquill, 2016-04-17:
@namelivia - I just had the exact same experience. 

2016-04-07_2016-03-18-octopi-jessie-lite-0.13.0.zip - Installs and works properly (RPi3) - although I needed to manually edit my wpa_supplicant

https://storage.googleapis.com/octoprint/octopi-jessie-lite-0.13.0_RC1.zip - After install has no web server running.

markwal, 2016-04-18:
@guysoft I verified that these reports are accurate.  The current RC1 (2016-04-09_2016-03-18-octopi-jessie-lite-0.13.0.zip) is just a bare jessie lite image.  It looks like the octopi script didn't run.  You can tell from the nightlies page because the images with no OctoPrint in them are 306.9MB (04-09 and 04-06).  The octopi script puts the image at about 475Mb.  You might want to use 04-08 or 04-18 instead.

guysoft, 2016-04-19:
Confirmed, there was a problem with that build (hash mismatch in debian installs), using now nightly from 2016-04-19

Please try the new RC2 at:
https://github.com/guysoft/OctoPi/issues/224

closed 2016-04-19 by guysoft

dschinsky, 2016-04-27:
trying this with the latest build.... after putting in the WIFI info... the system just reboots continuously. then finally stops with both the green and red light solid.

dschinsky, 2016-04-27:
also I wanted to know if there was any support in the 0.13 for the LCD touchscreen interface?

dschinsky, 2016-04-27:
to confirm it looks like it reboots as it is initializing the the WIFI

guysoft, 2016-04-27:
@dschinsky this issue is closed, please use RC2