guysoft/OctoPi #224: OctoPi 0.13.0 RC2 status

guysoft, 2016-04-19:
Please try the release candidate.
Download it at:
https://storage.googleapis.com/octoprint/octopi-jessie-lite-0.13.0_RC2.zip
md5:
https://storage.googleapis.com/octoprint/octopi-jessie-lite-0.13.0_RC2.zip.md5

Bananapi M1 (New variants expected at 0.14.0):
https://storage.googleapis.com/octoprint/octopi-bpi-r1-0.13.0_RC1.zip
MD5:
https://storage.googleapis.com/octoprint/octopi-bpi-r1-0.13.0_RC1.zip.md5

philphilphil, 2016-04-19:
Download link seems broken: Anonymous callers do not have storage.objects.get access to object octoprint/octopi-jessie-lite-0.13.0_RC2.zip.

guysoft, 2016-04-20:
Permission error, fixed

nicolinux, 2016-04-20:
Quick question, does the RC include the wifi fix for the Raspberry Pi 3?

guysoft, 2016-04-20:
Yes it should include the wifi fix, we waited to release because of it.

philphilphil, 2016-04-20:
I tested it today on a brand new Raspi 3. So far everything works without any issues. I changed the octopi-network.txt, installed some Plugins (Telegram, TouchUi) and run 10h of prints with no issues.

One thing to keep in mind: if you use Wi-fi channel 12 or 13 you need to enable them via raspi-config (Internationalisation Options > Change Wi-fi Country). They seem blocked by default.

If you need me to test something specific let me know.

dschinsky, 2016-04-27:
will this version support the lcd interface?

guysoft, 2016-04-27:
OctoPiPanel is supplied with OctoPi, but support of specific screens is not maintained at the moment, unless someone wants to (I have no time).

guysoft, 2016-05-01:
Released https://github.com/guysoft/OctoPi/releases/tag/0.13.0

closed 2016-05-01 by guysoft