guysoft/OctoPi #225: Acces octopi over the internet, port forwarding

SHurkmans, 2016-04-26:
I want to acces my octopi over the internet, when I port forward my printer, my octopi dissapears in my network list and I cannot acces the printer until I delete the port forwarding. When I set DMZ via the router everything works, but I don't want to do that because they say that it is not so secure.

I tried port forwarding on port 80 and 5000, but both don't seem to work

I don't understand a lot of DNS and port forwarding so maybe that's the issue

[image]

guysoft, 2016-04-28:
Some routers have problems with port forwarding, what they do is behave differently inside the LAN and WAN. When you set up your port 80, I'd try accessing the printer remotely and seeing if it works, it might be it works on the WAN, but not in LAN.
Apart from that I can't really know since this is not really an OctoPi issue.

alberto2000, 2016-05-02:
You could always use tunnelling via services like for example ngrok: www.ngrok.com - which works flawlessly in my case and there is a free plan which is perfectly fine for certain use cases

jerm, 2016-08-08:
Beware making your printer on publicly accessible. They're notoriously horribly insecure.

DarkSeverance, 2017-06-08:
Does your printer use port 80? If your printer is bound to port 80, then you'll want to use a different port for your OctoPi. For example, use port 1234 for OctoPi, then 4321 for Printer (if you can set it for your printer). If you haven't already, make sure you aren't using a default username/password. If you tend to have Octopi accessible from outside your LAN, then I'd suggest changing your default username from "pi" to something else entirely.

wjsteele, 2017-12-22:
Take a look at the Polar3D cloud plug-in for OctiPrint.  It's allows you secure remote access to your OctoPi or OctoPrint printer.

guysoft, 2019-07-02:
Closing, see https://octoprint.org/blog/2018/09/03/safe-remote-access/

closed 2019-07-02 by guysoft