guysoft/OctoPi #227: Raspberry pi camera v2

GarrethX, 2016-05-10:
Hello,

I hope it's the right place to post this. There is a new camera module for raspberry pi, which gives the possibility to shoot photos in QHD (3280×2464). However using this resolution throught mjpeg streamer is not possible - only HD is allowed.
It would be great if octoprint gave a choice for the user to use the v2 camera at full resolution for shooting timelapses. For this I assume that we should not use mjpeg streamer but native rpi function: camera.capture(). 
Obviously we can't use mjpeg and native functions at the same time for the same camera, however it is already possible to have mjpeg set-up for e.g. cheap usb webcam for printer monitoring, and we would just need a second raspi camera for shooting stills in high quality for timelapses.

Would it be possible?

lelandg, 2016-05-10:
I think I know this one, so I'm going to jump in here. The *camera* supports that higher resolution, yes. However, the OS does *not yet* support it! So best you can do is resolution of the old camera... until they update it, which I heard is planned within "a few months" (of the release date, IIRC).

So it won't be possible before that. ;-)

GarrethX, 2016-05-11:
Ah... so I've made a rookie mistake and assumed wrongly, that te problem is the mjpeg streamer and not in the OS (because who would sell official camera for $30 if it's not supported out of the box for out of the box pi3)... 
HOWEVER even if the camera are you sure that the mjpegstreamer will allow to use QHD?

guysoft, 2016-05-11:
As @GarrethX figured. The problem is indeed that mjpg-streamer streamer needs to be updated.
OctoPi uses a fork of mjpg-streamer that supports the picam, its assailable at:
https://github.com/jacksonliam/mjpg-streamer

@jacksonliam I see you have no issue tracker there. So it would be appreciated if you can update here if you got the time to develop this. Or if you need any access to hardware.

@lelandg We are all doing this in our spare time for the community. Usually when a solution exists its available via the nightly builds built on a machine I keep in my house 24/7. If you want to speed up shipping you are welcome to join us.

GarrethX, 2016-05-11:
@guysoft : If the mjpg-streamer can't support this camera, then maybe we could revise my first idea - not use the mjpg for shooting stills for timelapse? There is a python library using the rpi camera, allowing also for some image modifications like WB, rotation, ISO and many others. This way we could have normal cheap usb camera for fast streaming using mjpg (for this 640x480 or even 320x240 is enough), and python modules for awesome quality timelapses.

Of course assuming that pi will handle this camera, but I've found already, that it is possible to do rpi-ugrade, which would install new kerner which apparently supports 8MPX camera

jacksonliam, 2016-05-11:
Mjpg-streamer supports all that. I've had it running at resolutions above 1808p before on the old camera to use the full 5MP sensor. Rotation, ISO etc are supported, see: https://github.com/jacksonliam/mjpg-streamer/blob/master/mjpg-streamer-experimental/plugins/input_raspicam/README.md

You might be hitting up the limitations of the Raspberry Pi video mjpeg encoder if you're trying to do 8MP video. Try the usestills flag and a low FPS (e.g. 5) should enable you to use the full sensor. 

Obviously make sure you're pi is upto date. apt-get update && apt-get upgrade should be enough. 

If you want to use two cameras in theory you should be able to run two instances of mjpg-streamer (on different ports) and put their respective stream and snapshot uris into octoprint. But I've not tested that.

foosel, 2016-05-11:
I'd be surprised if running two instances wouldn't work. I'd suggest setting the second instance up as the snapshot one, that way the stream url redirection in haproxy doesn't have to be changed.

OctoPrint doesn't care if snapshot URL and stream URL are from the same server or entirely separate ones. All it does with the snapshot URL is try to download a jpg file from it and save it to disk. So if you don't want to use mjpg-streamer, there's also nothing hindering you from developing a tiny http server that invokes whatever alternative library there might exist to capture the full 8MP image from the camera and dump it into the response body.

jacksonliam, 2016-05-11:
You could, Although a tiny http server which grabs frames jpeg from a camera is literally the definition of mjpg streamer 😀

GarrethX, 2016-05-11:
Ok, guess I found the way how to set-up two mjpeg streamers:
http://www.thedoble.com/3d-printing/running-two-webcams-with-octoprint-for-hd-timelapses/
I'll give it a shot on the weekend. 
I guess the advantage of the second mjpeg streamer would be that it will run in a separate thread which will be good in terms of performance, right?. And if this will not work well, I'll try to implement standard PiCamera library, however I'm not yet sure how, but I'll figure this out:)

GarrethX, 2016-05-11:
Ok, today I've tested few things - without sudo rpi-update you can't go above FHD.
But after this command, magic happens - it works like a documentation says (not sure though what other things got broken in the process:D ).
For stream you can only have FHD, checked. 
For -shootstills you can use QHD 3280 x 2464 

I'll have the usb camera on the weekend, so I'll try to use two mjpg streamers like in the link above and we will see how it goes. If anyone would be interested, I'd post here the result, if not - you can close this topic

jstefanop, 2016-06-10:
this is exactly what i was looking for...any progress on this front? Maybe someone can create a octoprint plug in that will allow setting the snapshot URL to a separate instance of mjpg-streamer that shoots HD stills for HD time lapses. 

Does anyone know if two instances of mjpg-streamer can access the rpi cam at the same time? (or if the built in rpi cam API is accessible while the streamer is running.

jacksonliam, 2016-06-10:
@jstefanop what progress were you looking for? Raspberry Pi Camera V2 is already supported and you can already capture full resolution frames with mjpg-streamer. Octoprint already supports seperate snapshot and stream URIs.

Two applications cannot access the Pi Camera at the same time. Are you wanting HD stills and a low res stream from one camera? 

I don't think the Pis encoder can output multiple streams at once, at least not with the MMAL libs. Best you can do is scale differently for the encoder and display output.  Might be able to switch output resolution between frames in usestills mode to create two 'psudostreams', but each would be max 4 fps.

The Pi 3 and perhaps Pi 2 might have enough grunt to resize HD jpegs in software on-the-fly at a reasonable framerate.

foosel, 2017-02-28:
Closing as requested by OP.

closed 2017-02-28 by foosel